raulmur/ORB_SLAM2 #513: How can i use a usb webcam for orbslam2?

hesamira, 2018-01-23:
My lap top find my usb webcam.
I changed different parameters in usb_cam-test.launch and usb_cam_node.cpp . I mean, I used video1 instead of video0. In addition, I changed different values. I also made some changes in mono_tum.cc code, but i could not get video and see it. 

Please, guide me.

shridharkini6, 2018-01-23:
@hesamira
when launch file is running list the topic using `$rostopic list`, choose the proper topic and embed it in your c code.

hesamira, 2018-01-23:
I observed a green screen with coloured lines, but i could not saw the video clearly.

AlejandroSilvestri, 2018-01-29:
@hesamira,

You don't see your video clearly?  But you do see your video.  May be your video is there, beneath the green lines.

During initialization, ORB-SLAM2 display green lines showing candidates matches.  But usually you can see the grayscale image from camera with some green lines on it.

hesamira, 2018-01-29:
Unfortunately, I didn't see my video clearly. I also tested usb_cam-test.launch and usb_cam_node.cpp with various image formats such as yuyv, uyvy, mjpeg, yuvmono10, rgb24, but I didn't get any positive result.

mobiiin, 2021-02-28:
im sure you found the solution by now. but for others who are struggling with this:
you have to change the line 64 of the "ros_mono.cc" to this:
`ros::Subscriber sub = nodeHandler.subscribe("/usb_cam/image_raw", 1, &ImageGrabber::GrabImage,&igb);`
`~/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_mono.cc`
and then build the ros package by naivgating to ORB_SLAM2 folder and `./build_ros.sh`
tada