raulmur/ORB_SLAM2 #515: Fail to complie Examples/ROS/ORB_SLAM2,help

cdlwhm1217096231, 2018-01-26:
when i use "chmod +x build_ros.sh ./build_ros.sh",the following erro will be find,can you help me ,thank you!
/usr/bin/ld: CMakeFiles/Stereo.dir/src/ros_stereo.cc.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
/usr/lib/i386-linux-gnu/libboost_system.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/Stereo.dir/build.make:216: recipe for target '../Stereo' failed
make[2]: *** [../Stereo] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

WafeiBaBa, 2018-01-28:
I encountered the same problem. @cdlwhm1217096231

cdlwhm1217096231, 2018-01-28:
you can see this here   http://blog.csdn.net/cdlwhm1217096231/article/details/79177189

cdlwhm1217096231, 2018-01-28:
you can see this here http://blog.csdn.net/cdlwhm1217096231/article/details/79177189  @WafeiBaBa

phillip-s, 2018-01-29:
This PR https://github.com/raulmur/ORB_SLAM2/pull/507 might fix it(?)

cdlwhm1217096231, 2018-01-29:
I don't understand your means,sorry

phillip-s, 2018-01-29:
Try https://github.com/raulmur/ORB_SLAM2/pull/507

ebimor, 2018-03-08:
Thanks philip-s!

miaoxw, 2018-04-04:
#507 works perfectly. Thanks for the information from @phillip-s .