raulmur/ORB_SLAM2 #516: In localization mode, How can I load information about keyframes?

apeoej333, 2018-01-29:
Hello,

I have a question in localization mode.

In localization mode, We need information(Bow vector) about keyframes.
Because We find detect relocalization candidates by comparing Bow vector of current frame with that of keyframes.

I think that System::SaveKeyFrameTrajectoryTUM is just function that saves keyframe's pose and in rgbd_tum.cc, LoadImages is just function that load RGB-D images.

But I can't find code that save and load information about keyframes. 

Please, guide me.
Thank you very much.

AlejandroSilvestri, 2018-01-29:
@apeoej333 ,

There isn't.   You should write your own function to save keyframes' info you require.

apeoej333, 2018-01-30:
@AlejandroSilvestri 
Thank you for your reply.

If I only use Bow Vector(not using covisibility graph) to detect relocalization candidates, 
Do I need to save ID, Bow vector, mnRelocQuery, mnRelocWords, mnRelocScore of each keyframe??

Sincerely,

AlejandroSilvestri, 2018-01-31:
@apeoej333 

I believe you only need to save this keyframe's members in order to keep BoW info:

- mnId, unique keyframe's id
- mBowVec, vector of weighted BoWs if I recall correctly
- mFeatVec, linking BoWs with mappoints if I recall correctly

You'll also need mappoints to proceed with orb-slam2 relocalization.