raulmur/ORB_SLAM2 #521: Segmentation fault occurs in extreme cases

leiothrix0, 2018-02-12:
----Hi, I'm testing a video from a pair of driving recorders, and sometimes the program crashes with a segmentation fault. I've already located the cause, but I still need some confirmation on this bug, Thanks in advance.

**Background**
----The direct reason of this problem is solarization, in the video, some frames of the video had solarization, which is actually an extreme case. Another thing is that the program does not crash everytime.

**Scenirio**
----Frame1 and Frame2 are having solarization, Frame3 and Frame4 are normal frame. Thus we had a lost on Frame1, and Frame2 is the first frame after resetting. And the program crashes on frame4, which is the third frame after resetting 

**Possible Reason**
----Please take a look at:
https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L544
----From`Line 544` to `Line 551`

>        mLastFrame = Frame(mCurrentFrame);
>        mnLastKeyFrameId=mCurrentFrame.mnId;
>        mpLastKeyFrame = pKFini;
>
>        mvpLocalKeyFrames.push_back(pKFini);
>        mvpLocalMapPoints=mpMap->GetAllMapPoints();
>        mpReferenceKF = pKFini;
>        mCurrentFrame.mpReferenceKF = pKFini;

----`mCurrentFrame` is assigned to `mLastFrame` before `pKFini` is assigned to its `mpReferenceKF`, I don't know whether it's a mistake or intended.
----No matter what, if `mCurrentFrame.mpReferenceKF` equals `NULL`,  `mLastFrame.mpReferenceKF` equals `NULL`. Mark this and we'll continue, please take a look at:
https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L307 
----From`Line 307` to `Line 316`
>        if(mVelocity.empty() || mCurrentFrame.mnId<mnLastRelocFrameId+2)
>        {
>            bOK = TrackReferenceKeyFrame();
>        }
>        else
>        {
>            bOK = TrackWithMotionModel();
>            if(!bOK)
>                bOK = TrackReferenceKeyFrame();
>        }

----for Frame4, since it's the third frame after resetting, `mCurrentFrame.mnId<mnLastRelocFrameId+2` equals `false` ,and somehow `mVelocity.empty()` is `false`, then we will call `TrackWithMotionModel()`: 
https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L867
----please take a look at `Line 873`, it calls `UpdateLastFrame()`:
>    UpdateLastFrame();

----And in `UpdateLastFrame()` 
https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L801
----From `Line 804` to `Line 807`:
>    KeyFrame* pRef = mLastFrame.mpReferenceKF;
>    cv::Mat Tlr = mlRelativeFramePoses.back();
>
>    mLastFrame.SetPose(Tlr*pRef->GetPose());

----If somehow, `mLastFrame.mpReferenceKF` hasn't been changed after initialization, `pRef` equals `NULL` and `pRef->GetPose()` will finally caused a segmentation fault. 

**Possible Solution**
----I've listed what I got when I debug this problem, I'm not very familiar with the whole logic, thus I'm not very sure about it, intuitively speaking, I think if we place:
>        mLastFrame = Frame(mCurrentFrame);
after
>        mCurrentFrame.mpReferenceKF = pKFini;
this bug may be fixed. However, if this is intended, I think the problem is why the workflow enters `TrackWithMotionModel()` when `mLastFrame.mpReferenceKF` hasn't been assigned to a 'non-NULL' pointer. The pre-condition for `TrackWithMotionModel()` is for the third frame after resetting, `mVelocity` is not empty.
----Thanks in advance, and if needed, I can provide any data for this problem. Also, If you have anything unclear, feel free to ask me.

ronir121, 2018-12-11:
Is this solution working? I encounter the same seg fault

ns15417, 2020-01-09:
@ronir121 i am not sure whether you still need the answer. i encountered the same question and solved it by changing the location of  mLastFrame = Frame(mCurrentFrame); and change it after line mCurrentFrame.mpReferenceKF = pKFini; so that the reference frame when updateLastFrame() is not NULL