raulmur/ORB_SLAM2 #526: saving/loading maps

hesamira, 2018-02-21:
Hello,
I want to save map and load the data in ORB_SLAM2. I have performed the changes in ORBSLAM2 (I have made some change to System.cc, System.h, and ros_mono.cc.). 

**For System.cc, I added the following code:**

void System::SaveMapPoints(const string &filename) {
    cout << endl << "Saving map points to " << filename << " ..." << endl;

    vector<MapPoint*> vpMPs = mpMap->GetAllMapPoints();

    // Transform all keyframes so that the first keyframe is at the origin.
    // After a loop closure the first keyframe might not be at the origin.
    ofstream f;
    f.open(filename.c_str());
    f << fixed;

    for(size_t i=0; i<vpMPs.size(); i++) {
        MapPoint* pMP = vpMPs[i];

        if(pMP->isBad())
            continue;

        cv::Mat MPPositions = pMP->GetWorldPos();

        f << setprecision(7) << " " << MPPositions.at<float>(0) << " " << MPPositions.at<float>(1) << " " << MPPositions.at<float>(2) << endl;
    }

    f.close();
    cout << endl << "Map Points saved!" << endl;

}

**For System.h, I added the following code:**

void SaveMapPoints(const string &filename);

**For ros_mono.cc, I added the following code:**

SLAM.SaveMapPoints("MapPointsSave.txt");

The MapPointsSave.txt contains all the map points in terms of global locations, i.e., xyz coordinate. But I could not save a map and load the data. I also have created a package for pcd  (http://pointclouds.org/documentation/tutorials/writing_pcd.php) which I can convert the saved map points in .txt file into .pcd file. I could not run it. I have tested it by: 
$ ./pcd_write

Please offer help.
Thanks.

AlejandroSilvestri, 2018-02-27:
@hesamira ,

What do you mean when you say "I could not save a map"?

Does your code abort with errors?  Does MapPointsSave.txt generates properly?

hesamira, 2018-03-04:
Dear @AlejandroSilvestri ,
I have done the above changes in my code, but I could not save the map.
I have not faced with any error, but I could not save the map. So, MapPointsSave.txt has not been generated.

AlejandroSilvestri, 2018-03-05:
@hesamira,

How do you call SLAM.SaveMapPoints("MapPointsSave.txt")?

Are you sure this method is invoked at all?

hesamira, 2018-03-10:
@AlejandroSilvestri ,
I am not sure. I have used this link:
https://github.com/raulmur/ORB_SLAM/issues/5
Can you suggest the better code for save map and load the data in orbslam2? I have used different codes such as poine 's code and math 's code, but I could not success.

AlejandroSilvestri, 2018-03-11:
@hesamira ,

There are many ways around to save maps in orb.slam2, but any of them well implemented and documented, perhaps with the exception of poine's.

Make sure you call your savemap in some point of orb.slam2's code.  Maybe at the end, near [SaveKeyFrameTrajectoryTUM](https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/Monocular/mono_tum.cc#L123), for example.

Put a console mark (cout << "Saving!" << endl;)  as a way to verify your function has been invoked.

hesamira, 2018-03-13:
@AlejandroSilvestri ,
I have used your method. But I could not save the map. I have tried a lot.
I do not know where is the problem. :(
your answer is very important for me!
Please, guide me.

hesamira, 2018-03-23:
@AlejandroSilvestri ,
Hi, 
I have used MathewDenny ' code <https://github.com/mathewdenny> . When running the ros example, I got a problem.

sami@sami:~/orbslam2_catkin_ws$ source devel/setup.bash
sami@sami:~/orbslam2_catkin_ws$ rosrun ORB_SLAM2 Mono /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml

Usage: rosrun ORB_SLAM2 Mono path_to_vocabulary path_to_settings

I have checked my ROS master. I have checked my path. There are no errors. based on : https://github.com/raulmur/ORB_SLAM2/issues/43
I think that there is problem in ros_mono.cc file. Is it correct? I do not know where is the problem. I have tried a lot.
your answer is very important to me. I need to save map and load the data.

Please guide me.
Thanks.

rahulsharma11, 2019-02-07:
Hi @hesamira ,
I checked this and able to save the map.txt file.

AlejandroSilvestri, 2019-02-07:
Hi @rahulsharma11  , did you miss a link?  What did you check?

@hesamira , sorry I missed your post.  I can't help you though.  I'm not familiar with ROS nor have experience with MathewDenny's map save.