raulmur/ORB_SLAM2 #527: Cmake: OpenCV > 2.4.3 not found. , but via pip: opencv-contrib-python (3.4.0.12) is installed

bubbelMCU, 2018-02-25:
hello,  not sure what shall I do?  
cmake cannot find opencv, which I installed for python2 and 3 via pip and the version is:
opencv-contrib-python (3.4.0.12)

How to set path to opencv pip package correctly? 
Or do I need to recompile opencv from source?

Error message by Cmake:
```
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:35 (message):
  OpenCV > 2.4.3 not found.
```

douxiao, 2018-06-14:
https://blog.csdn.net/gxsheen/article/details/79090454

ishanivyas, 2019-09-24:
Having same issue with  OpenCV > 2.4.3 not found.

LongruiDong, 2019-11-03:
> https://blog.csdn.net/gxsheen/article/details/79090454

It WORKS!!!

Brian-ZhenLiu, 2020-08-10:
> https://blog.csdn.net/gxsheen/article/details/79090454

This really works.

Wingmore, 2020-08-24:
Could someone provide/explain the steps from the blog please? I don't understand and I've searched every other answer on the web

Wingmore, 2020-08-26:
nvm, I got it - had to add my openCV directory to  Thirdparty/DBoW2/CMakeLists.txt 

i.e. had to find my openCV directory. For me it was in /usr/local/include/opencv4.
Then I just added `SET (OpenCV_DIR /usr/local/include/opencv4)` to my CMakeLists.txt file.

ratomatrix, 2021-01-09:
Same error message for me in January 2021.
In my case, the issue was that I just installed the latest OpenCV (as of January 2021 - OpenCV 4.5.1), but versions above OpenCV 3 are not accepted. So I downloaded and installed the latest version starting with 3 (3.4.13), and this error message has gone.
(Unfortunately, I personally have other error messages now :) But at least this one seems to be solved.)

Emilianocb, 2022-10-09:
> nvm, I got it - had to add my openCV directory to Thirdparty/DBoW2/CMakeLists.txt
> 
> i.e. had to find my openCV directory. For me it was in /usr/local/include/opencv4. Then I just added `SET (OpenCV_DIR /usr/local/include/opencv4)` to my CMakeLists.txt file.

i can't added 'SET (OpenCV_DIR /usr/local/include/opencv4)' because the file are only to read,  I tried with chmod but it's the same

DANIELHARDLY, 2022-11-23:
> i can't added 'SET (OpenCV_DIR /usr/local/include/opencv4)' because the file are only to read, I tried with chmod but it's the same

use sudo to edit CMakeLists.txt
if you are using vim type  <sudo vim CMakeLists.txt>

mgrallos, 2023-03-04:
> SET (OpenCV_DIR /usr/local/include/opencv4)
where do I put it? I did already add it but same error occurs

supermice, 2025-03-11:
邮件已收到！ ***@***.***

lx-r, 2025-03-11:
您好，来信已收到，谢谢！

shuannibro, 2025-03-11:
你好，谢谢你的支持，我们已收到你的文件