raulmur/ORB_SLAM2 #529: question about function Tracking::UpdateLocalKeyFrames()

PetWorm, 2018-02-28:
does the "break;" necessary in line 1328? 
which break the mvpLocalKeyFrames updation once add a parent KF is added into mvpLocalKeyFrames.

Zander-yang, 2018-03-05:
I think the condition coulb be satisfied easily. And maybe it's a clerical error. Function UpdateLocalKeyFrames          is used to get potential Key Frames and limit the number of keyframes(80). So >80 is the final break condition. When the size of mvpLocalKeyFrames is less than 80, finding a not included parent frame should not break the circulation.

PetWorm, 2018-03-05:
@Zander-yang 
i do agree with the logic u replied at the beginning, 
i'm just wondering if this break is a necessity, for mvpLocalKeyFrames do not want to keep too much parent KFs
我一开始也是这么想的，我只是好奇是不是作者真的需要这么写来让mvpLocalKeyFrames不维护太多的父关键帧，