raulmur/ORB_SLAM2 #530: CMake 3+ not supported

a83282485, 2018-03-05:
The C++11 capability check always fails with MSVC (2015 and 2017).

The lines following "# Check C++11 or C++0x support" represent the old (i.e. pre CMake 3.x) style of checking for compiler or language features. It does not respect the command line flags of MSVC.

Here is the full error you will get:

```
The C compiler identification is MSVC 19.12.25835.0
The CXX compiler identification is MSVC 19.12.25835.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x86/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x86/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x86/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x86/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Build type: Release
Performing Test COMPILER_SUPPORTS_CXX11
Performing Test COMPILER_SUPPORTS_CXX11 - Failed
Performing Test COMPILER_SUPPORTS_CXX0X
Performing Test COMPILER_SUPPORTS_CXX0X - Failed
CMake Error at CMakeLists.txt:26 (message):
  The compiler C:/Program Files (x86)/Microsoft Visual
  Studio/2017/Community/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x86/cl.exe has
  no C++11 support.  Please use a different C++ compiler.
```

antonyleme, 2018-05-16:
I'm having the same problem. Did you fix it? How did you do that?

aksufatih, 2019-03-28:
Hi,
I am getting this error. How can I solve this problem? Could you please share the changed lines in the CMakeLists.txt? I have no experience on CMake and couldn't find a way to make it work.
Thank you

navalsheth, 2019-05-02:
Same Problem here it seems,just says 
cl.exe has
   The compiler C:/Program Files (x86)/Microsoft Visual
  Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe has
  no C++11 support.  Please use a different C++ compiler.

jaafar95, 2020-01-22:
you can try to replace: