raulmur/ORB_SLAM2 #532: failed  to building ORB_SLAM2

cdlwhm1217096231, 2018-03-06:
when I have finished ORB_SLAM2, I want to build Examples/ROS/ORB_SLAM2,the following errors will be generate，please help me,thank you !

cdl@cdl-Inspiron-5421:~/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ make
Scanning dependencies of target rospack_genmsg_libexe
[  0%] Built target rospack_genmsg_libexe
Scanning dependencies of target rosbuild_precompile
[  0%] Built target rosbuild_precompile
Scanning dependencies of target Stereo
[ 11%] Building CXX object CMakeFiles/Stereo.dir/src/ros_stereo.cc.o
make[2]: *** No rule to make target 'opencv_calib3d-NOTFOUND', needed by '../Stereo'。 停止。
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

hhyh, 2018-03-06:
Did cmake report any error before you run 'make'?

cdlwhm1217096231, 2018-03-06:
cmake can not find any errors

cdlwhm1217096231, 2018-03-06:
run 'cmake' not find any errors,just when I run 'make',the above errors will be generate

hhyh, 2018-03-06:
How you install the opencv3?
Build from source or install the binary from ubuntu repo / ros repo?

cdlwhm1217096231, 2018-03-06:
I install opencv2.4.11 and build from source

sunjiayu12, 2018-03-15:
Seems that your opencv doesn't have calib3d module. Try reinstall opencv.

cdlwhm1217096231, 2018-03-20:
when I am running ORB_SLAM2/ros_mono.cc,but the screen appear "TRYING TO INITIALIZE",i cannot find any reason,please help me,thank you!!!@sunjiayu12