raulmur/ORB_SLAM2 #537: Odometry Ros Topic - geometry_msgs/Twist.msg

Andreluizfc, 2018-03-13:
Hi,

is there a way to publish [Twist Messages](http://docs.ros.org/api/geometry_msgs/html/msg/TwistWithCovariance.html) using ORB SLAM?

Reading this PR #102 and [this git Project](https://github.com/idsc-frazzoli/ardrone_testbed/blob/master/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_mono.cc) I am able to publish PoseWithCovariance messages. What I need is a way to publish the whole `/odom` topic with ORB SLAM. Thanks beforehand!

André Castro.

marcelinomalmeidan, 2018-04-03:
@Andreluizfc, ORBSLAM does not estimate angular/linear velocities. You would need to estimate those yourself (use a Kalman Filter, for instance) to get velocities.

marcelinomalmeidan, 2018-04-03:
If fact, it looks like ORB-SLAM2 do some rudimentary velocity estimation between two consecutive frames. You can find it in Tracking.cc, when they set the variable mVelocity = mCurrentFrame.mTcw*LastTwc

Andreluizfc, 2018-04-03:
Great! 

I solved the problem using and EKF with the IMU. Nevertheless I should take a look in the file you just said. Thanks @marcelinomalmeidan !

closed 2018-04-03 by Andreluizfc

marcelinomalmeidan, 2018-04-03:
I believe your solution should work better than what they have! They just do a "dirty derivative" between two images (visual odometry), without taking into account IMU information. 
The EKF should do both at the same time (implicit visual odometry + IMU integration)

nominator, 2018-05-17:
@Andreluizfc Did you combine ORB_SLAM's pose estimate with external IMU using EKF? How did you synchronize the IMU sensor data stream with ORB_SLAM's pose estimates before feeding these to EKF? Any help will be much appreciated. Thanks

Haydar1990, 2022-08-19:
> Great!
> 
> I solved the problem using and EKF with the IMU. Nevertheless I should take a look in the file you just said. Thanks @marcelinomalmeidan !

Hello, i have a question and i hope yoe can answer me, have you used the ekf_localization_node in the robot_localization pkg to estimate the velocity and then to publish the whole /odom Topic?

Thank you.