raulmur/ORB_SLAM2 #538: Why the ORB uses modified verison of g2o and DBow?

vigorbird, 2018-03-14:
The ORB author said that the DBow and g2o is the modified version.The modified version g2o and DBow is just part of original  code? Or the ORB author changes some codes in original g2o and DBow codes? If the author changes the codes , why he does so?The ORB author said that the DBow and g2o is the modified version.The modified version g2o and DBow is just part of original  code? Or the ORB author changes some codes in original g2o and DBow codes? If the author changes the codes , why he does so?

sunjiayu12, 2018-03-15:
See **[ORB-SLAM: a Versatile and Accurate Monocular SLAM System](http://webdiis.unizar.es/~raulmur/MurMontielTardosTRO15.pdf)**

vigorbird, 2018-03-18:
@sunjiayu12
我看过这篇文章了，但是没有找到对于g2o和dbow库修改的相关说明。文章里只是简要说明了使用了这两个库。

sunjiayu12, 2018-03-20:
"...The original DBoW2 took this overlapping into account, adding up the score of images that are close in time. This has the limitation of not including keyframes viewing the same place but inserted at a different time. Instead we group those keyframes that are connected in the covisibility graph. ..."
原来的DBoW应该是只算了时间相近的帧的分数，但是按作者的意思应该也把相隔一段时间但相似度很高的图像计算进去。
g2o的修改我也没找到诶，只知道用的g2o里的LM……

vigorbird, 2018-03-21:
@sunjiayu12 
thanks，I am trying to find changes in g2o.

vigorbird, 2018-05-10:
@sunjiayu12 你好我知道orb中的g2o和原始版本的有什么区别了。在原始版本的g2o中计算雅克比使用的是数值解而orb中的g2o计算雅克比是解析解。