raulmur/ORB_SLAM2 #541: Fail to complie ORB_SLAM2

hesamira, 2018-03-17:
Hello,
When I have compiled ORB_SLAM2 code, I have faced this error:

sami@sami:~/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ make -j
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
[rosbuild] Building package ORB_SLAM2
[rosbuild] Cached build flags older than manifests; calling rospack to get flags
-- Using CATKIN_DEVEL_PREFIX: /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel
-- Using CMAKE_PREFIX_PATH: /home/sami/orbslam2_catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/sami/orbslam2_catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
Build type: RelWithDebInfo
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0")
-- Boost version: 1.54.0
-- Found the following Boost libraries:
-- system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
Scanning dependencies of target rospack_genmsg_libexe
[ 0%] Built target rospack_genmsg_libexe
Scanning dependencies of target rosbuild_precompile
[ 0%] Built target rosbuild_precompile
Scanning dependencies of target Mono
Scanning dependencies of target MonoAR
Scanning dependencies of target RGBD
Scanning dependencies of target Stereo
[ 20%] [ 40%] [ 60%] Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o
Building CXX object CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o
Building CXX object CMakeFiles/Stereo.dir/src/ros_stereo.cc.o
[ 80%] [100%] Building CXX object CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o
Building CXX object CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o
Linking CXX executable ../Mono
Linking CXX executable ../RGBD
Linking CXX executable ../Stereo
CMakeFiles/Mono.dir/src/ros_mono.cc.o: In function main': /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_mono.cc:59: undefined reference toORB_SLAM2::System::System(std::string const&, std::string const&, ORB_SLAM2::System::eSensor, bool)'
collect2: error: ld returned 1 exit status
CMakeFiles/Stereo.dir/src/ros_stereo.cc.o: In function main': /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_stereo.cc:64: undefined reference toORB_SLAM2::System::System(std::string const&, std::string const&, ORB_SLAM2::System::eSensor, bool)'
CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o: In function main': /home/sami/orbslam2_catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_rgbd.cc:62: undefined reference toORB_SLAM2::System::System(std::string const&, std::string const&, ORB_SLAM2::System::eSensor, bool)'
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
make[2]: *** [../Mono] Error 1
make[2]: *** [../RGBD] Error 1
make[2]: *** [../Stereo] Error 1
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
make: *** [all] Error 2

your answer is very important to me. 
Please guide me.
Thanks.

vanessakat, 2018-05-11:
hello sir, i am trying to compile this Project in Windows 7, 64 bit on visual studio 2017 (IDE), i am having  lot of fatal errors, .
could do tell me what i can do to run this Project...
some of my errors are:
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4091	'': ignored on left of 'int' when no variable is declared	Project4	c:\orb_slam2-master (1)\orb_slam2-master\gcc-master\gcc\ginclude\stdint-gcc.h	62	
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4091	'': ignored on left of 'int' when no variable is declared	Project4	c:\orb_slam2-master (1)\orb_slam2-master\gcc-master\gcc\ginclude\stdint-gcc.h	73	
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4091	'': ignored on left of 'unsigned int' when no variable is declared	Project4	c:\orb_slam2-master (1)\orb_slam2-master\gcc-master\gcc\ginclude\stdint-gcc.h	76	
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4244	'argument': conversion from 'double' to 'const float', possible loss of data	Project4	c:\orb_slam2-master (1)\orb_slam2-master\thirdparty\g2o\g2o\types\types_six_dof_expmap.h	126

vanessakat, 2018-05-11:
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4244	'argument': conversion from 'double' to 'const float', possible loss of data	Project4	c:\orb_slam2-master (1)\orb_slam2-master\thirdparty\g2o\g2o\types\types_six_dof_expmap.h	126	
Error	C1003	error count exceeds 100; stopping compilation	Project4	c:\orb_slam2-master (1)\orb_slam2-master\gcc-master\gcc\ginclude\stdint-gcc.h	78	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C1021	invalid preprocessor command 'cmakedefine'	Project4	c:\orb_slam2-master (1)\orb_slam2-master\pangolin-master\pangolin-master\src\config.h.in	14	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'DBL_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1172	
Error	C2065	'FLT_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1109	
Error	C2065	'FLT_TRUE_MIN': undeclared identifier	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1109	
[... cut ...]
Error	C2131	expression did not evaluate to a constant	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1259	
Error	C2131	expression did not evaluate to a constant	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1260	
Error	C2131	expression did not evaluate to a constant	Project4	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.13.26128\include\limits	1261	
Error	C2131	expression did not evaluate to a constant	Project4	c:\program files (x86)\micro

hesamira, 2018-05-12:
Hi,
I have tested ORB_SLAM2 in linux (ubuntu 16.04).
I am sorry, I havenot tested in Windows.

vanessakat, 2018-05-12:
Thank you for the answer.
I will install Linux on my pc
To run it