raulmur/ORB_SLAM2 #542: Using ORBSLAM2 with depth image only?

salocinx, 2018-03-17:
Hi. Is it possible to use ORBSLAM2 with the depth image only? My application will run in dark rooms and therefore the RGB image will not contain lots of information. If it needs both, RGB image and the depth-map, I will have to check for an infrared option.

AlejandroSilvestri, 2018-03-19:
@salocinx , not possible, ORB-SLAM2 is essencially visual.

Infrared image should work.

I'm not into it, but I believe there must be some code out there for SLAM with depth map only.

salocinx, 2018-03-23:
@AlejandroSilvestri , thanks for your response. I did search for RGBD only SLAM solutions, but didn't found anything of interest. Today's depth sensors have still some serious disadvantages (limited range, black or mirror surfaces, direct sunlight), therefore I understand that most SLAM approaches use the depth information only as supplementary ingredient.

Okay I will then try to implement an infrared approach. Hopefully I will find a IR frequency that satisfy the Astra RGB camera and does not interfere with the IR projector of the depth sensor. The illumination uniformity could also emerge as a major issue.

AlejandroSilvestri, 2018-03-27:
@salocinx 

Please let me know about your findings, and good luck!