raulmur/ORB_SLAM2 #544: ORB-SLAM2: Current Frame:11704): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

gethubwy, 2018-03-19:
我使用了TUM数据集的rgbd_dataset_freiburg1_xyz/数据，我在编译完成ORB SLAM2的时候，进行运行该程序，使用的命令是：./Examples/Monocular/mono_tum  Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ../rgbd_dataset_freiburg1_xyz/
但是当程序运行时，我发现并不能正常的初始化、不能正常的结束该程序。
下面是我的程序的输出信息：
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

(ORB-SLAM2: Current Frame:11704): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(ORB-SLAM2: Current Frame:11704): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(ORB-SLAM2: Current Frame:11704): GLib-GIO-CRITICAL **: g_dbus_connection_get_unique_name: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
New Map created with 75 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New Map created with 95 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New Map created with 106 points
到这里，我看到最后一张画面了，显示界面变成了灰色，我不得不使用CTRL+C强制结束这个程序，我不知道哪里出现了错误，希望得到大家的帮助。万分感谢！

hzfGuardian, 2018-03-19:
I've got the same error about this... Hope for anyone's help too~

ll884856, 2018-03-24:
请问问题解决了吗？我也遇到了一样的问题。

hzfGuardian, 2018-03-24:
Maybe while calling pangolin, there exists deadlock..
maybe you can try to delete the final two rows in function System::Shutdown(), src/System.cc

if(mpViewer)
        pangolin::BindToContext("ORB-SLAM2: Map Viewer");

And then try again？

gethubwy, 2018-03-24:
YES! It did work ! I don't know why there exists deadlock.What is the reason for this?Can you teach me?
Thank you very much！ @hzfGuardian

gethubwy, 2018-03-24:
问题已经解决了，感谢@hzfGuardian 的帮助
 我想请问一下你@ll884856 的环境，确认一下我们是不是环境的关系导致的这个问题。
我的环境是：Ubuntu 16.04；安装的过程请看http://www.cnblogs.com/wongyi/p/8596084.html这是我整理特详细的过程。

hzfGuardian, 2018-03-24:
@gethubwy I've not found out the reason why there exists deadlock yet. I can just assure that in [https://github.com/stevenlovegrove/Pangolin/blob/master/src/display/display.cpp](url), function PangolinGl *FindContext(), the deadlock occurs when calling contexts_mutex.lock(). 
If you are still interested about it, maybe you can try debug steps in this function to see the output result.

ll884856, 2018-03-24:
@gethubwy 我发现我和你的问题还不太一样，看了你的安装过程发现是我之前少装了g2o和DBoW2，装好重新编译就跑通了，没有改system.cc，我也是16.04， 非常感谢你的帮助！

gethubwy, 2018-03-25:
谢谢您~~~ @hzfGuardian

fredlllll, 2018-04-02:
the application also randomly stops responding for me =/