raulmur/ORB_SLAM2 #545: Integrate external speed/steer values with ORB-SLAM

AntonioMarra, 2018-03-22:
Hi all,
for what i've learned since now, ORB-SLAM uses visual odometry values to estimate the camera pose, and this method can often be not really accurate.
I'd like to "improve" the slam by using external data (like speed and steering values of a robot) obtained from other sensors, my question is: is this integration possible with orb-slam and if yes, do you guys have some tips (or link to similar works like this) to start with?

AlejandroSilvestri, 2018-03-22:
@AntonioMarra ,

May be you can combine localization data from orb-slam2 and odometry from other sensors, with a Kalman filter.

You could also use that sensors data to improve the first camera pose approximation by improving the motion model.

On a higher level, I recommend reading Visual-Inertial Monocular SLAM with Map Reuse, when they combine orb-slam2 with IMU sensors.

AntonioMarra, 2018-03-23:
Hi and thanks @AlejandroSilvestri for the answer, 

I've read some of the code of the SLAM and i've found (correct me if i'm wrong) that the motion model is only used in the Track() function in Tracking.cc. 

This function uses a cv::Mat called mVelocity for setting the Pose of the current frame, with respect to the pose of the previous frame, in this way: mCurrentFrame.SetPose(mVelocity*mLastFrame.mTcw);
 
The mVelocity matrix is then updated later in the Track() function by using the pose just updated from in the current frame and some transformation of the pose of the previous frame, this way: 
mVelocity = mCurrentFrame.mTcw*LastTwc;

i've also understood that those arrays are 4x4 matrix but i have not understood yet what is the "meaning" of the values contained in the mTcw matrix (the pose of the frame) and in the mVelocity matrix (the motion model) 

My idea was to modify the mVelocity vector, using some fixed data obtained from the extern (like the exact speed of the camera motion), to make the motion model more accurate, since i think that the key of this motion model is mVelocity. It's this idea logic or am i totally wrong?

Last question, can someone explain me the exact meaning of the values in mVelocity and mTcw vectors?

AlejandroSilvestri, 2018-03-26:
@AntonioMarra 

You are right.

Tracking starts with a pose estimation from motion model, which  if I remember correctly is just uniform linear motion.

mTcw is a 4x4 pose matrix.  The pose is a combination of 3x3 rotation matrix and 3x1 position vector (a 3x4 matrix), in homogeneous coordinates (hence 4x4):

| [ R ] [t] |
| 0 0 0 1 |

In mTcw:
- m stands for member of object
- T stands for Transformation
- c stands for camera
- w stands for world

mTcw is the world pose in the camera reference.  mTwc is the camera pose in world reference.  One is the inverse of the other.

    mVelocity = mCurrentFrame.mTcw*LastTwc

means mVelocity is the actual camera pose expressed in the last camera pose's reference system.

wangshuang870039239, 2018-09-19:
@AlejandroSilvestri 
From "mVelocity = mCurrentFrame.mTcw*LastTwc",
If the mean of mVelocity is the last camera's pose expressed in the current camera reference system ? Maybe it will help you to understand "mCurrentFrame.SetPose(mVelocity*mLastFrame.mTcw)" well.