raulmur/ORB_SLAM2 #546: stop without saving the trajectory

ll884856, 2018-03-23:
Hi everyone,
I have built the ORB-SLAM2 and run the dataset successfully. But after running the dataset, the program just stops and the video turns grey like this screenshot.  It doesn't save the "KeyFrameTrajectory.txt" as well. How can I solve this problem? Thanks!

[image]

ll884856, 2018-03-24:
Problem has solved. It is because I didn't install g2o and DBoW2.

BoomFan, 2018-03-27:
HI, I have the same problem. 
Are you sure about this?
I already run `./build.sh`, shouldn't it install g2o and DBoW2 automatically?

ll884856, 2018-03-28:
You can try to run these commands and run ./build.sh 

sudo apt-get install libboost-dev
sudo apt-get install cmake  libeigen3-dev  libsuitesparse-dev qtdeclarative5-dev 
 qt5-qmake  libqglviewer-dev

They are dependencies of g2o and DBoW2.

BoomFan, 2018-03-28:
Thanks, but it doesn't work for me.
I installed all the dependencies you listed and I'm sure no error pops out when I `./build.sh`.
Unfortunately, after I run the demo code, the window still remain grey and it seems that the program has stuck there and cannot move on. Just like your situation, it doesn't save the "KeyFrameTrajectory.txt" as well.

BoomFan, 2018-03-28:
Actually, I followed issues547 and tried his solution. It fixes my problem. 
https://github.com/raulmur/ORB_SLAM2/issues/547

Now have new problems, I'll guess I'll just comment on issues 547 for my new problem.
Thanks for your help again.

yuqiag, 2018-04-27:
just delete the code"if(mpViewer)   pangolin::BindToContext("ORB-SLAM2: Map Viewer");"

georgebola, 2018-04-29:
In which file can we find that code ?

SouthEastleo, 2018-12-19:
> just delete the code"if(mpViewer) pangolin::BindToContext("ORB-SLAM2: Map Viewer");"

thanks,I have solved this problem throng your answer.