raulmur/ORB_SLAM2 #547: Shutdown problem in System.cc

kjwkch, 2018-03-26:
Hi.

I found shutdown fuction problem.
I think that mpViewer is not cleared.
So, program is killed by pangolin::BindToContext("ORB-SLAM2: Map Viewer") line on windows.
I added codes that clear mpViewer and have used.

`void System::Shutdown()

{

    mpLocalMapper->RequestFinish();

    mpLoopCloser->RequestFinish();

    if(mpViewer)

    {

        mpViewer->RequestFinish();

        while(!mpViewer->isFinished())

            usleep(5000);
        /* add code
        delete mpViewer;
        mpViewer = static_cast<Viewer*>(NULL);
        */
    }

    // Wait until all thread have effectively stopped

    while(!mpLocalMapper->isFinished() || !mpLoopCloser->isFinished() || mpLoopCloser->isRunningGBA())

    {

        usleep(5000);

    }

    if(mpViewer)  // <- always 'true'

        pangolin::BindToContext("ORB-SLAM2: Map Viewer");

}`

BoomFan, 2018-03-28:
Thanks! I was running into a shutdown problem and I tried your code. It works!
But somehow, after the trajectory is saved, I got errors like this:
```QObject::~QObject: Timers cannot be stopped from another thread
Segmentation fault (core dumped)```
Do you have the same issue?

kjwkch, 2018-03-28:
No. I think that the issue is occurred saving trajectory before ORB_SLAM2.Shutdown().
Call Shutdown first.

Below the code is 'System.h'
`// All threads will be requested to finish.
    // It waits until all threads have finished.
    // This function must be called before saving the trajectory.
    void Shutdown();

    // Save camera trajectory in the TUM RGB-D dataset format.
    // Only for stereo and RGB-D. This method does not work for monocular.
    // Call first Shutdown()
    // See format details at: http://vision.in.tum.de/data/datasets/rgbd-dataset
    void SaveTrajectoryTUM(const string &filename);

    // Save keyframe poses in the TUM RGB-D dataset format.
    // This method works for all sensor input.
    // Call first Shutdown()
    // See format details at: http://vision.in.tum.de/data/datasets/rgbd-dataset
    void SaveKeyFrameTrajectoryTUM(const string &filename);

    // Save camera trajectory in the KITTI dataset format.
    // Only for stereo and RGB-D. This method does not work for monocular.
    // Call first Shutdown()
    // See format details at: http://www.cvlibs.net/datasets/kitti/eval_odometry.php
    void SaveTrajectoryKITTI(const string &filename);`

BoomFan, 2018-03-28:
Here is my `mono_kitti.cc` file. It seems like `SaveKeyFrameTrajectoryTUM()` is at the very end of this code, right?
[image]

But I still got this error:
[image]

Do you think we have to add more to `void System::Shutdown()` function?

PengshuaiChai, 2018-03-28:
I followed your suggestion  and it worked.Then ,I have the same question with BoomFan .
[image]

dyz-zju, 2018-04-29:
QObject::~QObject: Timers cannot be stopped from another thread
Segmentation fault (core dumped)

I met same error as you after followed your suggestion. Do you find the solution?
@kjwkch @BoomFan @PengshuaiChai

dyz-zju, 2018-05-04:
I have solved the problem which is related to Pangolin branch, we need not to change the source code and just need to use a older branch of pangolin.

georgebola, 2018-05-05:
can you provide the branch of pangolin that worked for you ( link if possible)

dyz-zju, 2018-05-05:
you can provide me a email, I will send to you.@georgebola

georgebola, 2018-05-05:
georgensbis@gmail.com thank you.

ryanismert, 2018-05-08:
I had success just commenting out the call to ```BindToContext()```. Save trajectory was called fine, and no segfault.

joebedard, 2018-05-21:
I debugged `BindToContext()` and it looks like there is a mutex deadlock in Pangolin. The workaround suggested by @kjwkch also works for me. I think it is a better approach because (deleting mpViewer) should detach the viewer thread from the (OpenGL) context (see [Pangolin Issue 111](https://github.com/stevenlovegrove/Pangolin/issues/111) comment from stevenlovegrove). I suspect this is the intention of the the call to `BindToContext()`.

mayankamedhe, 2018-05-31:
@kjwkch My System.h file has the functions in the same order as posted by you. But I am still getting segmentation fault error. How can I solve that error?

caolk, 2018-05-31:
Thanks! I solve this problem according to your suggestion. @dyz-zju 
I found an old version of Pangolin at https://github.com/stevenlovegrove/Pangolin/tree/8b8b7b96adcf58ac2755dedd3f681fc512385af0
It works quite well.

ap108, 2018-07-09:
Hi, I tried the code edits to the present version of Pangolin, and the older version of Pangolin but I also get 
```
QObject::~QObject: Timers cannot be stopped from another thread
Segmentation fault (core dumped)
```

@kjwkch @BoomFan @PengshuaiChai 
Is there a solution?

kerolex, 2018-08-25:
Thanks @kjwkch. 

I compiled Pangolin from the master branch on the 9th of August and your solution works perfectly fine on my side with all files, e.g. trajectories, saved.