raulmur/ORB_SLAM2 #549: tum1 example output is flaky and incomplete (short:crap)

fredlllll, 2018-04-02:
ok so a lot of trouble with the tum1 example.
1) if viewer is enable it locks up at random points. so i disabled the viewer as im only interested in the resulting trajectory file... (also removed the waiting for next frame)
2) after processing trajectory file contains 30 images out of 798.... WAT?

apparently orb slam loses tracking very quickly. i get several of these on the terminal:

```
rgb/1305031106.175534.png
New Map created with 96 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
rgb/1305031106.211275.png
```

for an example that is an awful lot of problems. what is going wrong here?

i need the trajectory file to use the Map2DFusion software to generate an orthomosaic image from arbitrary drone imagery(all i have are jpgs with gps data). am i on the right track?

is SLAM.Shutdown(); stopping processing prematurely? i couldnt find anything else that would wait for processing to finish

MichaelGrupp, 2018-06-23:
> after processing trajectory file contains 30 images out of 798.... WAT?

You're probably looking at the keyframe poses. That would be expected.

fredlllll, 2018-06-24:
then how do i get the positions of the frames in between? i would need those