raulmur/ORB_SLAM2 #556: How to apply my calibration results to the YAML settings file?

salocinx, 2018-04-16:
Hi. I am working with Orbbec Astra Mini depth sensor and did the camera calibration according to the OpenCV checkboard calibration procedure. I got the following results:

```
[Left Camera Intrinsic]
578.457       0 346.077
      0 571.928 250.389
      0       0       1
[Right Camera Intrinsic]
517.723       0 337.738
      0 512.275 242.524
      0       0       1
[Right to Left Camera Rotate Matrix]
   0.999841 -0.00841257   0.0157468
 0.00861483    0.999881  -0.0128205
  -0.015637   0.0129542    0.999794
[Right to Left Camera Translate]
-24.8819 0.138074 -0.203578 
```

With the help of this tutorial and the depicted matrix, I can extract the `fx`, `fy`, `cx` and `cy` values for the left and right camera (left camera is the IR camera and right camera is RGB camera).

But how do I have to apply these values to my YAML file? Within the YAML file, there's only 1x {fx, fy, cx, cy} values to be defined:

```
# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 535.4
Camera.fy: 539.2
Camera.cx: 320.1
Camera.cy: 247.6

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.width: 640
Camera.height: 480
```

Shouldn't there be Camera.fx/.fy/.cx/.cy for both cameras?

Moreover, how do I have to extract the k1/k2/p1/p2/p3 parameters from the matrix I got from the calibration process (depicted above) ?

JajaIstso, 2018-04-18:
I have a similar question regarding the
**Camera.bf**  parameter in stereo usage. Understandable is **baseline *fx**. Is the baseline here in pixel or with a unit like mm? 

Thanks in advance.

YuJeHyeon, 2020-06-03:
Excuse me?
Is that left camera mean ir camera, right camera mean rgb camera on Astra Mini model?