raulmur/ORB_SLAM2 #562: Load a map from a PCD file and run ORB SLAM 2 in localization mode

anika001, 2018-04-26:
Hi everyone, 

First, I would like to thank you for sharing this nice work.

I am working on ORB SLAM 2 in monocular mode. I managed to store the map in a pcd file that I can view with pcl_viewer.

Now, I want to load this map to ORB SLAM 2 (to perform if possible visual odometry) and hence I should activate the localization mode, I would appreciate if you can give more explanations on how to run ORB SLAM 2 on localization mode and what should I modify in the code if possible. Also, do you have any idea on how to load the map to ORB SLAM 2 ?  

Thank you in advance

xmojiao, 2018-05-04:
maybe you can think about this answer[https://github.com/raulmur/ORB_SLAM2/issues/19](url).I have try Alkaid-Benetnash's  [github](https://github.com/Alkaid-Benetnash/ORB_SLAM2) ,and it is working

hesamira, 2018-05-07:
Dear @xmojiao,
I have tested Alkaid-Benetnash's github. But I coulden't save and load my map with ROS or without ROS. How could you save and load your map via Alkaid-Benetnash's github?

Please, guide me.
Thanks.

xmojiao, 2018-05-07:
@hesamira 
  yes, I can save and load map, then get camera Trajectory for the second bag file. In fact,  Alkaid-Benetnash only gave the Mono model. If you need RGBD model to get map, you must modify something in TUMx file and rgbd_tum.cc like mono_tum.cc.

bazmohammad, 2020-09-24:
How to save map for orbslam2 Ros Mono? Can someone please help me out this problem.