raulmur/ORB_SLAM2 #568: g2o internal compile error on Win 10, VS 17

JeffyOLOLO, 2018-05-18:
when i try compile g2o emerge internal compile error in a file **g2o/core/matrix_operations.h** in a template **axpy**. 
when i comment the template then the error emerge in a template **atxpy**. when i comment it also, then compilator can't find **std::max()** function.

i tried add more memory in swap space (#305), but it still not work for VS compilator.

maybe Windows requires somewhere its own inclusions somewhere instead Unix inclusions?

i'm using the latest Eigen version - 3.3.4. also, i try with 3.3.1-3.3.3

Visual Studio build output:
```
1>------ Build started: Project: g2o, Configuration: Release x64 ------
1>types_sba.cpp
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\hyper_graph.h(93): warning C4996: 'std::tr1': warning STL4002: The non-Standard std::tr1 namespace and TR1-only machinery are deprecated and will be REMOVED. You can define _SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING to acknowledge that you have received this warning.
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\unordered_map(1065): note: see declaration of 'std::tr1'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(62): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_sba.h(41): note: see reference to class template instantiation 'g2o::BaseVertex<3,Eigen::Vector3d>' being compiled
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(99): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(99): note: while compiling class template member function 'int g2o::BaseVertex<3,Eigen::Vector3d>::stackSize(void) const'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_sba.h(41): note: see reference to class template instantiation 'g2o::BaseVertex<3,Eigen::Vector3d>' being compiled
1>types_six_dof_expmap.cpp
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\hyper_graph.h(93): warning C4996: 'std::tr1': warning STL4002: The non-Standard std::tr1 namespace and TR1-only machinery are deprecated and will be REMOVED. You can define _SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING to acknowledge that you have received this warning.
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\unordered_map(1065): note: see declaration of 'std::tr1'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\robust_kernel.h(77): warning C4996: 'std::tr1': warning STL4002: The non-Standard std::tr1 namespace and TR1-only machinery are deprecated and will be REMOVED. You can define _SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING to acknowledge that you have received this warning.
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\functional(2683): note: see declaration of 'std::tr1'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(62): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_sba.h(41): note: see reference to class template instantiation 'g2o::BaseVertex<3,Eigen::Vector3d>' being compiled
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_binary_edge.h(59): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.h(80): note: see reference to class template instantiation 'g2o::BaseBinaryEdge<2,Eigen::Vector2d,g2o::VertexSBAPointXYZ,g2o::VertexSE3Expmap>' being compiled
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_binary_edge.h(60): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.h(126): warning C4244: 'argument': conversion from 'double' to 'const float', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.cpp(151): warning C4244: 'initializing': conversion from 'double' to 'float', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.cpp(151): warning C4244: 'initializing': conversion from 'double' to 'const float', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.cpp(300): warning C4244: 'initializing': conversion from 'double' to 'float', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.cpp(300): warning C4244: 'initializing': conversion from 'double' to 'const float', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(99): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(99): note: while compiling class template member function 'int g2o::BaseVertex<6,g2o::SE3Quat>::stackSize(void) const'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.h(59): note: see reference to class template instantiation 'g2o::BaseVertex<6,g2o::SE3Quat>' being compiled
1>types_seven_dof_expmap.cpp
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\hyper_graph.h(93): warning C4996: 'std::tr1': warning STL4002: The non-Standard std::tr1 namespace and TR1-only machinery are deprecated and will be REMOVED. You can define _SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING to acknowledge that you have received this warning.
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\unordered_map(1065): note: see declaration of 'std::tr1'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\robust_kernel.h(77): warning C4996: 'std::tr1': warning STL4002: The non-Standard std::tr1 namespace and TR1-only machinery are deprecated and will be REMOVED. You can define _SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING to acknowledge that you have received this warning.
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\functional(2683): note: see declaration of 'std::tr1'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_vertex.h(62): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_sba.h(41): note: see reference to class template instantiation 'g2o::BaseVertex<3,Eigen::Vector3d>' being compiled
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\core\base_binary_edge.h(59): warning C4996: 'Eigen::AlignedBit': was declared deprecated
1>c:\users\bakozaur\desktop\univer\diplom\scam\eigen-git-mirror\eigen\src\core\util\constants.h(166): note: see declaration of 'Eigen::AlignedBit'
1>c:\users\bakozaur\desktop\univer\diplom\scam\orb_slam2\thirdparty\g2o\g2o\types\types_six_dof_expmap.h(80): note: see reference to class template instantiation 'g2o::BaseBinaryEdge<2,Eigen::Vector2d,g2o::VertexSBAPointXYZ,g2o::VertexSE3Expmap>' being compiled
[... cut ...]
1>Please choose the Technical Support command on the Visual C++
1> Help menu, or open the Technical Support help file for more information
1>INTERNAL COMPILER ERROR in 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\CL.exe'
1>    Please choose the Technical Support command on the Visual C++
1>    Help menu, or open the Technical Support help file for more information
1>Done building project "g2o.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========
```

JeffyOLOLO, 2018-05-18:
i included `<algorithm>` where need for `std::max()`.
and i found and inserted a code of [`vasprintf()`](http://lists.llvm.org/pipermail/cfe-commits/Week-of-Mon-20130325/077069.html) and the build was succeeded.
but a commenting of not compilable code it isn't solution, isn't it?
```
int vasprintf( char **sptr, const char *__restrict fmt, va_list ap )
{
    *sptr = NULL;
    int count = vsnprintf( NULL, 0, fmt, ap ); // Query the buffer size required.
 if( count >= 0 ) {
     char* p = static_cast<char*>(malloc(count+1)); // Allocate memory for it.
     if ( p == NULL )
         return -1;
        if ( vsnprintf( p, count+1, fmt, ap ) == count ) // We should have used exactly what was required.
            *sptr = p;
        else { // Otherwise something is wrong, likely a bug in vsnprintf. If so free the memory and report the error.
            free(p);
            return -1;
        }
    }
    return count;
}
```

EnricoBeltramo, 2019-05-07:
> i included `<algorithm>` where need for `std::max()`.
> and i found and inserted a code of [`vasprintf()`](http://lists.llvm.org/pipermail/cfe-commits/Week-of-Mon-20130325/077069.html) and the build [...]

Can you post modified files? I'm having same issue and try to apply same changes, but without success