raulmur/ORB_SLAM2 #570: Pangolin Make error

sharwanram, 2018-05-28:
Scanning dependencies of target pangolin
[  1%] Building CXX object src/CMakeFiles/pangolin.dir/utils/file_extension.cpp.o
[  2%] Building CXX object src/CMakeFiles/pangolin.dir/utils/timer.cpp.o
[  3%] Building CXX object src/CMakeFiles/pangolin.dir/utils/sigstate.cpp.o
[  4%] Building CXX object src/CMakeFiles/pangolin.dir/utils/file_utils.cpp.o
[  5%] Building CXX object src/CMakeFiles/pangolin.dir/utils/threadedfilebuf.cpp.o
[  6%] Building CXX object src/CMakeFiles/pangolin.dir/utils/uri.cpp.o
[  7%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_zstd.cpp.o
[  8%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_raw.cpp.o
[  9%] Building CXX object src/CMakeFiles/pangolin.dir/image/pixel_format.cpp.o
[ 10%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_pango.cpp.o
[ 11%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_exr.cpp.o
[ 12%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_png.cpp.o
[ 13%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_jpg.cpp.o
[ 14%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_tga.cpp.o
[ 15%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_ppm.cpp.o
[ 16%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io.cpp.o
[ 17%] Building CXX object src/CMakeFiles/pangolin.dir/log/packet.cpp.o
[ 18%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream_reader.cpp.o
[ 19%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream.cpp.o
[ 20%] Building CXX object src/CMakeFiles/pangolin.dir/log/playback_session.cpp.o
[ 21%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream_writer.cpp.o
[ 22%] Building CXX object src/CMakeFiles/pangolin.dir/gl/gltexturecache.cpp.o
In file included from /home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.h:248:0,
                 from /home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gltexturecache.h:30,
                 from /home/gpslab/ORB_SLAM2/Pangolin/src/gl/gltexturecache.cpp:28:
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlTexture::CopyFrom(const pangolin::GlTexture&)':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:298:40: error: 'glCopyImageSubData' was not declared in this scope
                        width, height, 1);
                                        ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlRenderBuffer::Reinitialise(GLint, GLint, GLint)':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:464:42: error: 'glDeleteRenderbuffersEXT' was not declared in this scope
         glDeleteRenderbuffersEXT(1, &rbid);
                                          ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:469:35: error: 'glGenRenderbuffersEXT' was not declared in this scope
     glGenRenderbuffersEXT(1, &rbid);
                                   ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:470:52: error: 'glBindRenderbufferEXT' was not declared in this scope
     glBindRenderbufferEXT(GL_RENDERBUFFER_EXT, rbid);
                                                    ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:471:81: error: 'glRenderbufferStorageEXT' was not declared in this scope
  glRenderbufferStorageEXT(GL_RENDERBUFFER_EXT, internal_format, width, height);
                                                                              ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In destructor 'pangolin::GlRenderBuffer::~GlRenderBuffer()':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:479:42: error: 'glDeleteRenderbuffersEXT' was not declared in this scope
         glDeleteRenderbuffersEXT(1, &rbid);
                                          ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In destructor 'pangolin::GlFramebuffer::~GlFramebuffer()':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:528:41: error: 'glDeleteFramebuffersEXT' was not declared in this scope
         glDeleteFramebuffersEXT(1, &fbid);
                                         ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In constructor 'pangolin::GlFramebuffer::GlFramebuffer(pangolin::GlTexture&, pangolin::GlRenderBuffer&)':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:535:34: error: 'glGenFramebuffersEXT' was not declared in this scope
     glGenFramebuffersEXT(1, &fbid);
                                  ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In constructor 'pangolin::GlFramebuffer::GlFramebuffer(pangolin::GlTexture&, pangolin::GlTexture&, pangolin::GlRenderBuffer&)':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:544:34: error: 'glGenFramebuffersEXT' was not declared in this scope
     glGenFramebuffersEXT(1, &fbid);
                                  ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlFramebuffer::Bind() const':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:553:50: error: 'glBindFramebufferEXT' was not declared in this scope
     glBindFramebufferEXT(GL_FRAMEBUFFER_EXT, fbid);
                                                  ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:555:52: error: 'glDrawBuffers' was not declared in this scope
     glDrawBuffers( attachments, attachment_buffers );
                                                    ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlFramebuffer::Reinitialise()':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:562:41: error: 'glDeleteFramebuffersEXT' was not declared in this scope
         glDeleteFramebuffersEXT(1, &fbid);
                                         ^
[... 42 lines cut ...]
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:691:31: error: 'glDeleteBuffers' was not declared in this scope
         glDeleteBuffers(1, &bo);
                               ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlBuffer::Bind() const':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:697:33: error: 'glBindBuffer' was not declared in this scope
     glBindBuffer(buffer_type, bo);
                                 ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlBuffer::Unbind() const':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:702:32: error: 'glBindBuffer' was not declared in this scope
     glBindBuffer(buffer_type, 0);
                                ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlBuffer::Upload(const GLvoid*, GLsizeiptr, GLintptr)':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:708:56: error: 'glBufferSubData' was not declared in this scope
     glBufferSubData(buffer_type,offset,size_bytes, data);
                                                        ^
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp: In member function 'void pangolin::GlBuffer::Download(GLvoid*, GLsizeiptr, GLintptr) const':
/home/gpslab/ORB_SLAM2/Pangolin/include/pangolin/gl/gl.hpp:715:61: error: 'glGetBufferSubData' was not declared in this scope
     glGetBufferSubData(buffer_type, offset, size_bytes, data);
                                                             ^
src/CMakeFiles/pangolin.dir/build.make:566: recipe for target 'src/CMakeFiles/pangolin.dir/gl/gltexturecache.cpp.o' failed
make[2]: *** [src/CMakeFiles/pangolin.dir/gl/gltexturecache.cpp.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/pangolin.dir/all' failed
make[1]: *** [src/CMakeFiles/pangolin.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

JeffyOLOLO, 2018-05-30:
Check your OpenGL instalation.

zengtotty, 2018-06-01:
I have the same problem, but after I install glew, it still have the same problem.
sudo apt-get install libglew-dev
sudo apt-get install glew-utils
sudo apt-get install libglew1.13
sudo apt-get install libglewmx-dev
@sharwanram  do you have any suggestion on this. I am also using ubuntu 16.04

gyt290811, 2022-06-17:
> 

do you have the details of solving this problem?

gitouni, 2022-07-08:
> I have the same problem, but after I install glew, it still have the same problem. sudo apt-get install libglew-dev sudo apt-get install glew-utils sudo apt-get install libglew1.13 sudo apt-get install libglewmx-dev @sharwanram do you have any suggestion on this. I am also using ubuntu 16.04

I have solved this problem by using pangolin-v0.6. Please try a earlier pangolin version at [https://github.com/stevenlovegrove/Pangolin/releases](https://github.com/stevenlovegrove/Pangolin/releases)