raulmur/ORB_SLAM2 #572: ORB SLAM 2 hangs after the frames gets over

nik13, 2018-05-30:
I have being using orb slam2 on TUM datasets. It starts normally but hangs after the images get over. I tested the same thing on my video, same behaviour. I have to kill the task to stop it.

I tried installing all the required libraries again, but it didn't help. 
Using Ubuntu 16.04

[image]

weidongguo, 2018-05-30:
The frames are the inputs.
What do you expect when the program gets no more inputs? 
When the program has nothing to work with, it stops.

Or I might not be understanding your question.

nik13, 2018-05-30:
Its the same mono_tum program in examples. It do not stops. Just hangs when the frames are over.

davide200294, 2018-05-30:
I've got the same error with the same dataset. Images are over, but ORB-SLAM has to do few more operations including save the KeyframeTrajectory.txt file so it should still run

mayankamedhe, 2018-05-31:
I solved the problem by referring to #547

closed 2018-05-31 by nik13