raulmur/ORB_SLAM2 #573: Error while build

JiangCe0810, 2018-05-30:
I build opencv-3.2.0 successfully (it seemed) on ubuntu 16.04. 
But when building,the following error is generated.Please help me,thank you.
 /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFIsTiled@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference toTIFFOpen@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFReadEncodedStrip@LIBTIFF_4.0' 1> 
/usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference toTIFFSetField@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFWriteScanline@LIBTIFF_4.0'1>
 /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference toTIFFGetField@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFScanlineSize@LIBTIFF_4.0' 1>
/usr/lib/libopencv_highgui.so.2.4: undefined reference toTIFFSetWarningHandler@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFSetErrorHandler@LIBTIFF_4.0' 1> 
/usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference toTIFFReadEncodedTile@LIBTIFF_4.0' 1> 
/usr/local/lib/libopencv_imgcodecs.so.3.2.0:undefined reference to TIFFReadRGBATile@LIBTIFF_4.0' 1> 
/usr/local/lib/libopencv_imgcodecs.so.3.2.0:undefined reference toTIFFClose@LIBTIFF_4.0' 1> /usr/local/lib/libopencv_imgcodecs.so.3.2.0: undefined reference to TIFFRGBAImageOK@LIBTIFF_4.0' 1>
/usr/local/lib/libopencv_imgcodecs.so.3.2.0:undefined reference toTIFFReadRGBAStrip@LIBTIFF_4.0

weidongguo, 2018-05-30:
when you run pkg-config --libs --cflags opencv, what do you see?

Do you see tiff there?

TIFF is a module in OpenCV, so make sure you have checked TIFF=ON when building opencv

Here's a detailed discussion about your error https://github.com/BVLC/caffe/issues/4436

JiangCe0810, 2018-05-30:
I saw the method and I choose to delete the path of Anaconda,but the new error appears.
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/jiangce/Pangolin/include/pangolin/gl/gl.h:39,
                 from /home/jiangce/Pangolin/include/pangolin/pangolin.h:33,
                 from /home/jiangce/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/jiangce/ORB_SLAM2/include/Viewer.h:26,
                 from /home/jiangce/ORB_SLAM2/include/Tracking.h:28,
                 from /home/jiangce/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/jiangce/ORB_SLAM2/src/FrameDrawer.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

DingYu95, 2018-06-13:
as suggested [BVLC/caffe/issues/4436](https://github.com/BVLC/caffe/issues/4436#issuecomment-283632019), no need for rebuild OpenCV, just remove anaconda from PATH, might solve it.