raulmur/ORB_SLAM2 #575: Crash after loading the last image

davide200294, 2018-05-30:
Hello everybody,

today I installed ORB-SLAM2 and I tried to use the Monocular TUM Dataset example as proposed on your github page. It works until ORB-SLAM loads the last image, then it crash. Because of this crash I cannot obtain the KeyFrameTrajectory.txt file. Does anyone know how to fix this problem? 

P.S. I tried to use the RGB-D version and the problem is still present.

P.P.S. When I lunched the build.sh it returned few warnings about Eigen (which I installed, obviously). Can it be due to Eigen? The warnings was about binder2nd class.

I also attach the screenshot of the crash (so how it shows)
[image]

and the log on terminal:

"ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

init done
opengl support available
New Map created with 88 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New Map created with 94 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New Map created with 130 points"

Thanks in advance.

Regards

nik13, 2018-05-30:
I'm also having the same issue from a long time. No solutions till yet.

craymichael, 2018-06-09:
I too am having this issue

craymichael, 2018-06-09:
Update, see #547 for solution (downgrade pangolin)