raulmur/ORB_SLAM2 #576: rosrun ORB_SLAM2 Stereo -> Segmentation fault (core dumped)

saesaem, 2018-06-01:
I follow this web page : https://github.com/raulmur/ORB_SLAM2 
I met problem ....

root@258bf54d0a5e:~/ORB_SLAM2/Examples/ROS# rosrun ORB_SLAM2 MonoAR Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
Segmentation fault (core dumped)

How to solve it .... ?

guchangjun, 2018-06-24:
you should install eigen 3.3.4

gaunthan, 2018-12-21:
> you should install eigen 3.3.4

Still have this problem...

gaunthan, 2018-12-25:
> > you should install eigen 3.3.4
> 
> I use eigen3.2.10 and also get the error, how can i solve it?

Are you running the command on a remote terminal? If that's the case, run it locally.

imjrl, 2022-07-07:
Have you solved it?
I met the same problem.

Ultramarine1939-syujie, 2023-07-15:
I met the same problem.
does anyone have sovled this ?

gaunthan, 2023-07-15:
您的来信我已收到，我会尽快回复。