raulmur/ORB_SLAM2 #578: Can the ORB SLAM be run with low fps? My data set has an fps of 1. How can I tweak the code to work for 1 fps?

samidh15, 2018-06-03:


AlejandroSilvestri, 2018-06-04:
In the main loop, for example [here](https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/ROS/ORB_SLAM2/src/ros_mono.cc#L93), the code call SLAM to process image (TrackMonocular for example).  You can call it one time each second, to run in real time.

But if you process images in no real time, you don't have to tweak anything.

samidh15, 2018-06-05:
Thanks for that clarification. So I am not.processing in real time. I just
know that my data set is shot at 1 fps. But if I run ORB with this set, and
in the config file mention that fps is 1, ORB runs but gives me no output.
What am I missing here?

AlejandroSilvestri, 2018-06-06:
@samidh15 

Fps has no impact on orb-slam2 in no real time.  If I recall right, the only purpose of fps in configuration is to calculate a timestamp on each image of the sequence.  ORB-SLAM2 doesn't use it.

I do not know why you don't have output.  Did you try putting 30 fps in the config file?  I don't think it would be any different.

ORB-SLAM2 need a minimum of 100 feature correspondences with enough camera displacement to initialize the system and build the initial map.  So you usually need a lot more correspondences between frames.

And remember, it is imperative to have a really good calibration on your camera: focal point, central point and distortion parameters.

behretv, 2018-12-31:
@samidh15 

I had a similar problem and indeed changing the fps affected the tracking behaviour. Setting the fps too high or too low decreased the chances for a successful tracking. I guess the fps are used to estimate the area for possible correspondences. Additionally, the tracking fails if the changes between two frames are too abrupt. Thus, I "fixed" the problem by driving slower. But I don't know if that is an option for you.

WeiYongqiang55, 2019-08-07:
> Thanks for that clarification. So I am not processing in real time. I just know that my data set is shot at 1 fps. But if I run ORB with this set, and in the config file mention that fps is 1, ORB runs but gives me no output. What am I missing here? Additionally, can it happen that owing to the low frame rate, the feature correspondence between successive images is low, and ORB requires a minimum number of features to work? That is what I have been noticing in the logs (number of correspondences are sometimes are as low as 1 or 2.
> […](#)

I meet the same thing as you said ,my dataset fps is 1 ,and it stacked in the init step,do you solve this problem??