raulmur/ORB_SLAM2 #579: Pangolin make error

Zico2017, 2018-06-03:
Hello, everyone!
when I run "cmake --build .", I met some errors, I paste the compile error below, is anyone know the reason and how to solve it? Thank you in advance.

yezi-zhu@yezi-zhu:~/Downloads/Pangolin/build$ cmake --build .
Scanning dependencies of target pangolin
[  1%] Building CXX object src/CMakeFiles/pangolin.dir/utils/threadedfilebuf.cpp.o
[  2%] Building CXX object src/CMakeFiles/pangolin.dir/utils/timer.cpp.o
[  3%] Building CXX object src/CMakeFiles/pangolin.dir/utils/sigstate.cpp.o
[  4%] Building CXX object src/CMakeFiles/pangolin.dir/utils/file_utils.cpp.o
[  5%] Building CXX object src/CMakeFiles/pangolin.dir/utils/uri.cpp.o
[  6%] Building CXX object src/CMakeFiles/pangolin.dir/utils/file_extension.cpp.o
[  7%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_jpg.cpp.o
[  8%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io.cpp.o
[  9%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_png.cpp.o
[ 10%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_tga.cpp.o
[ 11%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_raw.cpp.o
[ 12%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_ppm.cpp.o
[ 13%] Building CXX object src/CMakeFiles/pangolin.dir/image/pixel_format.cpp.o
[ 14%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_exr.cpp.o
[ 15%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_zstd.cpp.o
[ 16%] Building CXX object src/CMakeFiles/pangolin.dir/image/image_io_pango.cpp.o
[ 17%] Building CXX object src/CMakeFiles/pangolin.dir/log/packet.cpp.o
[ 18%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream_reader.cpp.o
[ 19%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream.cpp.o
[ 20%] Building CXX object src/CMakeFiles/pangolin.dir/log/playback_session.cpp.o
[ 21%] Building CXX object src/CMakeFiles/pangolin.dir/log/packetstream_writer.cpp.o
[ 22%] Building CXX object src/CMakeFiles/pangolin.dir/gl/glfont.cpp.o
In file included from /home/yezi-zhu/Downloads/Pangolin/src/gl/glfont.cpp:44:0:
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2495:15: warning: ‘int stbtt_BakeFontBitmap(const unsigned char*, int, float, unsigned char*, int, int, int, int, stbtt_bakedchar*)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_BakeFontBitmap(const unsigned char *data, int offset,  // f
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2540:16: warning: ‘void stbtt_GetBakedQuad(stbtt_bakedchar*, int, int, int, float*, float*, stbtt_aligned_quad*, int)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_GetBakedQuad(stbtt_bakedchar *chardata, int pw, int ph, in
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2643:15: warning: ‘int stbtt_PackBegin(stbtt_pack_context*, unsigned char*, int, int, int, int, void*)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_PackBegin(stbtt_pack_context *spc, unsigned char *pixels, i
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2674:16: warning: ‘void stbtt_PackEnd(stbtt_pack_context*)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_PackEnd  (stbtt_pack_context *spc)
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2978:15: warning: ‘int stbtt_PackFontRange(stbtt_pack_context*, unsigned char*, int, float, int, int, stbtt_packedchar*)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_PackFontRange(stbtt_pack_context *spc, unsigned char *fontd
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2680:16: warning: ‘void stbtt_PackSetOversampling(stbtt_pack_context*, unsigned int, unsigned int)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_PackSetOversampling(stbtt_pack_context *spc, unsigned int 
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:2990:16: warning: ‘void stbtt_GetPackedQuad(stbtt_packedchar*, int, int, int, float*, float*, stbtt_aligned_quad*, int)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_GetPackedQuad(stbtt_packedchar *chardata, int pw, int ph, 
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1510:16: warning: ‘void stbtt_GetFontVMetrics(const stbtt_fontinfo*, int*, int*, int*)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_GetFontVMetrics(const stbtt_fontinfo *info, int *ascent, i
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1517:16: warning: ‘void stbtt_GetFontBoundingBox(const stbtt_fontinfo*, int*, int*, int*, int*)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_GetFontBoundingBox(const stbtt_fontinfo *info, int *x0, in
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1505:16: warning: ‘void stbtt_GetCodepointHMetrics(const stbtt_fontinfo*, int, int*, int*)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_GetCodepointHMetrics(const stbtt_fontinfo *info, int codep
                ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1200:15: warning: ‘int stbtt_GetCodepointBox(const stbtt_fontinfo*, int, int*, int*, int*, int*)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_GetCodepointBox(const stbtt_fontinfo *info, int codepoint, 
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1205:15: warning: ‘int stbtt_IsGlyphEmpty(const stbtt_fontinfo*, int)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_IsGlyphEmpty(const stbtt_fontinfo *info, int glyph_index)
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1156:15: warning: ‘int stbtt_GetCodepointShape(const stbtt_fontinfo*, int, stbtt_vertex**)’ defined but not used [-Wunused-function]
 STBTT_DEF int stbtt_GetCodepointShape(const stbtt_fontinfo *info, int unicode_c
               ^
/home/yezi-zhu/Downloads/Pangolin/src/gl/stb_truetype.h:1537:16: warning: ‘void stbtt_FreeShape(const stbtt_fontinfo*, stbtt_vertex*)’ defined but not used [-Wunused-function]
 STBTT_DEF void stbtt_FreeShape(const stbtt_fontinfo *info, stbtt_vertex *v)
                ^
[... 56 lines cut ...]
[ 57%] Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/unpack.cpp.o
[ 58%] Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/join.cpp.o
[ 59%] Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/merge.cpp.o
[ 60%] Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/json.cpp.o
[ 61%] Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/thread.cpp.o
[ 62%] Building CXX object src/CMakeFiles/pangolin.dir/tools/video_viewer.cpp.o
[ 63%] Building CXX object src/CMakeFiles/pangolin.dir/display/device/display_wayland.cpp.o
/home/yezi-zhu/Downloads/Pangolin/src/display/device/display_wayland.cpp:600:5: error: ‘pointer_handle_frame’ was not declared in this scope
     pointer_handle_frame,
     ^
/home/yezi-zhu/Downloads/Pangolin/src/display/device/display_wayland.cpp:601:5: error: ‘pointer_handle_axis_source’ was not declared in this scope
     pointer_handle_axis_source,
     ^
/home/yezi-zhu/Downloads/Pangolin/src/display/device/display_wayland.cpp:602:5: error: ‘pointer_handle_axis_stop’ was not declared in this scope
     pointer_handle_axis_stop,
     ^
/home/yezi-zhu/Downloads/Pangolin/src/display/device/display_wayland.cpp:603:5: error: ‘pointer_handle_axis_discrete’ was not declared in this scope
     pointer_handle_axis_discrete,
     ^
src/CMakeFiles/pangolin.dir/build.make:1574: recipe for target 'src/CMakeFiles/pangolin.dir/display/device/display_wayland.cpp.o' failed
make[2]: *** [src/CMakeFiles/pangolin.dir/display/device/display_wayland.cpp.o] Error 1
CMakeFiles/Makefile2:136: recipe for target 'src/CMakeFiles/pangolin.dir/all' failed
make[1]: *** [src/CMakeFiles/pangolin.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Zico2017, 2018-06-03:
The reason is virtual memory exhausted. When I set the last line of build.sh to
make -j1 
it solved

The issue can be closed

closed 2018-06-03 by Zico2017

alcatrazappu, 2018-12-17:
I have a similar problem, but using make -j1 did not solve it, in fact didnt have any effect at all
here is the error it shows with and without modification.

xt18t004@godavari:~/ORB_SLAM2$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:                                                                             148 (message):
  Could NOT find CUDA: Found unsuitable version "10.0", but required is exact
  version "9.0" (found /usr/local/cuda)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_F                                                                             AILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_                                                                             args)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:86 (find_package)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:105 (find_host_package)
  CMakeLists.txt:27 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/xt18t004/ORB_SLAM2/Thirdparty/DBoW2/build/CMakeFiles/CMakeOutput                                                                             .log".
See also "/home/xt18t004/ORB_SLAM2/Thirdparty/DBoW2/build/CMakeFiles/CMakeError.                                                                             log".
make: *** No targets specified and no makefile found.  Stop.
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xt18t004/ORB_SLAM2/Thirdparty/g2o/bui                                                                             ld
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CUDA: /usr/local/cuda-9.0 (found suitable exact version "9.0")
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xt18t004/ORB_SLAM2/build
Scanning dependencies of target ORB_SLAM2
make[2]: *** No rule to make target '/home/xt18t004/Pangolin/build/src/libpangol                                                                             in.so', needed by '../lib/libORB_SLAM2.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
[ 15%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBextractor.cc.o
[ 18%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o
[ 25%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBmatcher.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Map.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o
[ 37%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o
[ 40%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o
[ 43%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
[ 46%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Sim3Solver.cc.o
[ 50%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o
[ 53%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
[ 56%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o
[ 59%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' faile                                                                             d
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
xt18t004@godavari:~/ORB_SLAM2$

ishanivyas, 2019-09-26:
+1 Having the same issue.