raulmur/ORB_SLAM2 #581: Trying to plot the odometry values (in absolute values) on a map

freerunner-18, 2018-06-04:
@raulmur 
Hello sir,

We are working on an autonomous car and we are using orb slam2 for the purpose of localization as one of the methods of localization and while comparing it with the ground truth which is waht we are getting from a gps. There is a slight rotation and a bit of scaling which we are unable to trace. We are using the zed stereo camera and stereo_kitti along with rectified images from zed ros wrapper. The following image is the example of the plot. 
[image]

AlejandroSilvestri, 2018-06-04:
I believe GPS is plotted in green and ORB-SLAM2 in red.

ORB-SLAM2 adopts an arbitrary world reference, with origin where initialization took place.

You should manually align orb-slam2 output to the map in order to compare it with gps.

MichaelGrupp, 2018-06-23:
This can be caused by various things. For example:
* misalignment of visual SLAM tracking frame with respect to GPS coordinates
* rotational drift
* scale drift or wrong scale due to bad camera calibration
* ...
---
If it helps, you can check out this SLAM evaluation package: https://github.com/MichaelGrupp/evo

It includes tools to (scale) align trajectories and to measure absolute or relative differences.