raulmur/ORB_SLAM2 #583: OpenCV Matrix to calibration file

antonyleme, 2018-06-05:
Hello. I am trying to make my own sequence and calibrate my camera and make the yaml file, so I've already calibrated the camera with opencv and I got this result:

```
%YAML:1.0
---
image_width:1280
image_height:720
camera_matrix: !!opencv-matrix
 rows: 3
 cols: 3
 dt: f
 data: [1038.87, 0, 639.47, 0, 1043.82, 355.943, 0, 0, 1 ]
distortion_coefficients: !!opencv-matrix
 rows: 1
 cols: 5
 dt: f
 data: [0.273952, -1.30094, -0.00045381, -0.000324383, 1.88161 ]
```
How do I apply this to orb slam? I saw the yaml file from kitti dataset and I have no idea of where to apply this values. Can someone help me? This is from kitti.yaml:

```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 718.856
Camera.fy: 718.856
Camera.cx: 607.1928
Camera.cy: 185.2157

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

# Camera frames per second 
Camera.fps: 10.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.1
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 1
Viewer.PointSize:2
Viewer.CameraSize: 0.15
Viewer.CameraLineWidth: 2
Viewer.ViewpointX: 0
Viewer.ViewpointY: -10
Viewer.ViewpointZ: -0.1
Viewer.ViewpointF: 2000
```

antonyleme, 2018-06-06:
I've solved this with what here in brazil we call "chutometro". I tested the values until it works well. To who have the same doubt, I used according to TUM1.yaml as following:

```
%YAML:1.0
---
image_width:1280
image_height:720
camera_matrix: !!opencv-matrix
 rows: 3
 cols: 3
 dt: f
 data: [(Camera.fx)1038.87, 0, (Camera.cx)639.47, 0, (Camera.fy)1043.82, (Camera.cy)355.943, 0, 0, 1 ]
distortion_coefficients: !!opencv-matrix
 rows: 1
 cols: 5
 dt: f
 data: [(Camera.k1)0.273952, (Camera.k2)-1.30094, (Camera.p1)-0.00045381, (Camera.p1)-0.000324383, (Camera.k3)1.88161 ]
```
I dont know if it is completely right, but it worked very well actually and for now it's enough for me. If it is wrong and someone could explain the right way, I would appreciate that. Thanks

hm-tested, 2018-07-18:
Basically, OpenCV solves for the camera and distortion matrices which are:
```
Camera Matrix     Distortion Matrix
| fx 0 cx |       | k1 k2 p1 p2 k3 |
| 0 fy cy |
| 0  0  1 |
```
How it outputs it though is where the camera matrix is just one row and it simply tells you that its a 3x3 matrix. Your guess and check method seems to have gotten it exactly right though!