raulmur/ORB_SLAM2 #592: ./Build.sh causes system crash

Richard-Owen-3DI, 2018-06-19:
Hello,

I'm trying to install ORB_SLAM2 on a TOSHIBA Satellite running Ubuntu 16.04

I have compiled Pangolin successfully from source, and the file libpangolin.so can be found at /home/[user name]/packages/Pangolin/build/src

 Following the install instructions though and executing the build.sh script causes the system to install up to about 60%, and then totally freezes so I have to power cycle the machine (also meaning i am unable to capture a screen shot)

following the processes of 

mkdir build
cd build 
cmake .. 

gives me the warning of: 

"CMake Warning (dev) at CMakeLists.txt:87 (target_link_libraries):
  Cannot specify link libraries for target "pangolin" which is not built by
  this project.

  CMake does not support this but it used to work accidentally and is being
  allowed for compatibility.

  Policy CMP0016 is not set: target_link_libraries() reports error if its
  only argument is not a target.  Run "cmake --help-policy CMP0016" for
  policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it."

attempting to compile from this point will give me:
"
[ 56%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM2.so'. Stop.
make[2]: *** Waiting for unfinished jobs....
[ 59%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
"

I'm unsure how to progress, Should I modify the CMakefile.txt? Is Pangolin not as well installed as I believe it to be? 

Would really appreciate any help.

Many thanks.

saikrn112, 2018-06-19:
Hi Richard, 
This might be due to number of jobs `make` is taking to compile. Depending on the number of cores available `make` will try to parallelize the compilation process. So system might freeze if `make` is using all the cores for compilation. (although not sure of why it might happen). I decreased the compilation jobs to 3 using `make -j 3` (since I have 4 cores in my PC) and it compiled properly. 

Please try to decrease the compilation jobs and check if the error exists.

Richard-Owen-3DI, 2018-06-20:
Hello, 

Thank you for your response. I decreased the number of compilation jobs to 1. The system no longer freezes, which is progress of sorts, but it still fails to compile. The error I get is as follows:

"[ 59%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM2.so'. Stop.
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2"

Above in the generation and configuration by cmake I am getting the same message as before:

"CMake Warning (dev) at CMakeLists.txt:87 (target_link_libraries):
 Cannot specify link libraries for target "pangolin" which is not built by
 this project."

As well as hundreds of OpenCV errors, an example of which can be seen below:

"error: #error "OpenCV 4.x+ requires enabled C++11 support"
 #  error "OpenCV 4.x+ requires enabled C++11 support""

Thanks once again for your help

Richard-Owen-3DI, 2018-06-20:
***UPDATE***

I have now compiled DBoW2, which required an older version of OpenCV to install properly. 

Now when trying to compile ORB_SLAM2 I get the following error:

"[ 68%] Linking CXX executable ../Examples/Monocular/mono_euroc
/usr/bin/ld: warning: libopencv_core.so.3.1, needed by ../Thirdparty/DBoW2/lib/libDBoW2.so, may conflict with libopencv_core.so.3.4
CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o:(.data+0x0): undefined reference to `vtable for pangolin::HandlerScroll'
CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o:(.data+0x8): undefined reference to `vtable for pangolin::Handler'
CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o: In function `pangolin::GlTexture::Reinitialise(int, int, int, bool, int, unsigned int, unsigned int, void*)':
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x4a): undefined reference to `glGenTextures'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x57): undefined reference to `glBindTexture'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x7c): undefined reference to `glTexImage2D'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x99): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xad): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xc1): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xd5): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xda): undefined reference to `glGetError'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x108): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x11c): undefined reference to `glTexParameteri'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x12b): undefined reference to `pangolin::glErrorString(unsigned int)'
mono_euroc.cc:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x19d): undefined reference to `glDeleteTextures'
CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o: In function `pangolin::GlTexture::~GlTexture()':
mono_euroc.cc:(.text._ZN8pangolin9GlTextureD2Ev[_ZN8pangolin9GlTextureD5Ev]+0x15): undefined reference to `pangolin::ShouldQuit()'
mono_euroc.cc:(.text._ZN8pangolin9GlTextureD2Ev[_ZN8pangolin9GlTextureD5Ev]+0x28): undefined reference to `glDeleteTextures'
CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o: In function `pangolin::GlTexture::~GlTexture()':
mono_euroc.cc:(.text._ZN8pangolin9GlTextureD0Ev[_ZN8pangolin9GlTextureD5Ev]+0x21): undefined reference to `pangolin::ShouldQuit()'
mono_euroc.cc:(.text._ZN8pangolin9GlTextureD0Ev[_ZN8pangolin9GlTextureD5Ev]+0x33): undefined reference to `glDeleteTextures'
../lib/libORB_SLAM2.so: undefined reference to `glClearColor'
../lib/libORB_SLAM2.so: undefined reference to `glColor3f'
../lib/libORB_SLAM2.so: undefined reference to `glMultMatrixd'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, pangolin::Params const&)'
../lib/libORB_SLAM2.so: undefined reference to `glClear'
../lib/libORB_SLAM2.so: undefined reference to `glEnable'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::BindToContext(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::VarState::I()'
../lib/libORB_SLAM2.so: undefined reference to `glEnd'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::FinishFrame()'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::OpenGlRenderState::Follow(pangolin::OpenGlMatrix const&, bool)'
../lib/libORB_SLAM2.so: undefined reference to `glBegin'
../lib/libORB_SLAM2.so: undefined reference to `glPopMatrix'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreateDisplay()'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::ProjectionMatrix(int, int, double, double, double, double, double, double)'
../lib/libORB_SLAM2.so: undefined reference to `glBlendFunc'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::View::SetHandler(pangolin::Handler*)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::Handler3D::Handler3D(pangolin::OpenGlRenderState&, pangolin::AxisDirection, float, float)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::OpenGlRenderState::SetModelViewMatrix(pangolin::OpenGlMatrix)'
../lib/libORB_SLAM2.so: undefined reference to `glVertex3f'
[... cut ...]
collect2: error: ld returned 1 exit status
CMakeFiles/mono_euroc.dir/build.make:115: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2"

So it seems like the compilation being able to find the Pangolin library is my only outstanding issue.

Tien94, 2018-07-02:
Hi, I also run ORB-SLAM2 version of Jiankai (https://github.com/Jiankai-Sun/ORB_SLAM2_Enhanced) on Ubuntu16.04LTS and got the issue: make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so'.

Then I solved it by manually copied the file libDBoW2.so (which I successfully built from another ORB-SLAM2 source (https://github.com/raulmur/ORB_SLAM2) ) to the respective folder /Thirdparty/DBoW2/lib of my current ORB-SLAM2  directory and built again. 

I attached the libDBoW2.so in case you'd like to try this way.
[libDBoW2.so.tar.gz](https://github.com/raulmur/ORB_SLAM2/files/2153726/libDBoW2.so.tar.gz)