raulmur/ORB_SLAM2 #603: ROS build error

ujjwalrathod007, 2018-07-02:
I want to build ROS node. I have set the environment variable as follows and also added it to .bashrc. Still I get the errrors as shown bellow. Anyone has idea?

export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2

`ujjval@ujjval-VPCEH18FG:~/ORB_SLAM2$ ./build_ros.sh
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is "/home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
`

saikrn112, 2018-07-02:
Hey Rathod, 
As the line
 `[rosbuild] rospack found package "ORB_SLAM2" at "", but the current
directory is "/home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2"`
 indicates your ros environment variable is not setup correctly. After exporting the ROS_PACKAGE_PATH cross verify if it is setup correctly by `echo $ROS_PACKAGE_PATH` 
also make sure that you are exporting and running the build_ros.sh in the same terminal.

ujjwalrathod007, 2018-07-02:
Hello, As far as I know ROS_PACKAGE_PATH should be where the packages are
I have set them like follows, still the error persists.
`ujjval@ujjval-VPCEH18FG:~/ORB_SLAM2$ echo $ROS_PACKAGE_PATH
/opt/ros/kinetic:/opt/ros/kinetic/share:home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2`

saikrn112, 2018-07-03:
Did you try `rospack find ORB_SLAM2` after setting the `ROS_PACKAGE_PATH` variable?

ujjwalrathod007, 2018-07-03:
I checked it but it could not find the ORB_SLAM2 package. So I cloned the SLAM2 library using git in my /opt/ros/kinetic/share folder. then it was able to find the package but the error was not resoved

I saw a similar issue in[https://github.com/raulmur/ORB_SLAM/issues/39](url)

ujjwalrathod007, 2018-07-04:
Following the steps shown on the web-page worked for me. closing
[https://gitlab.tu-berlin.de/breakdowncookie/ORB_SLAM2/tree/a0493c223e41c37032e5c9103f8b4724f1217663](url)

closed 2018-07-04 by ujjwalrathod007

SuperBigBrother, 2018-07-12:
Hi~ I'm facing the same error as same as yours~ But when I clicking the website, I see 404... Would you please give me some help~I will appreciate that!!

VitoLing, 2018-12-08:
did you have source .bashrc

VitoLing, 2018-12-08:
I found the way man ,maybe we are the same situation, don't let the "source /opt/ros/kinetic/setup.bash" as the last line in ~/.bashrc, it will change the $ROS_PACKAGE_PATH, make sure export ${ROS_PACKAGE_PATH}:home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2 as the last line

pankhurivanjani, 2019-01-07:
I am facing the same error but the link given is not opening.
Can someone please help in this?

kevinstudyprogram, 2019-09-10:
I also encountered this error, how can I do?

nhjydywd, 2021-08-12:
I just solved the same problem as you encountered.
The cause of the problem is that my `ROS_PACKAGE_PATH` is wrong. (Although I once thought it was correct)
So I strongly reconmend you to check  your `ROS_PACKAGE_PATH`.
`export ORB_SLAM_PATH={your orb_slam2 path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2`
`export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:${ORB_SLAM_PATH}`
run command `cd ${ORB_SLAM_PATH}`  in terminal to insure the path is correct.
and then you can run `./build_ros.sh`

Xu-Huitong, 2022-02-26:
sloved
echo "source /opt/ros/noetic/setup.bash" >> ~/.bashrc
source ~/.bashrc

tungtungyan, 2022-05-18:
> I just solved the same problem as you encountered. The cause of the problem is that my `ROS_PACKAGE_PATH` is wrong. (Although I once thought it was correct) So I strongly reconmend you to check your [...]

My result is this.
echo $ROS_PACKAGE_PATH
/opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS
How can I remove the medium part ? (/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS)

nhjydywd, 2022-05-18:
> My result is this. echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS How can I remove the medium part ? [...]

You can redefine the variable:
` export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS`

nhjydywd, 2022-05-18:
> > > > I just solved the same problem as you encountered. The cause of the problem is that my `ROS_PACKAGE_PATH` is wrong. (Although I once thought it was correct) So I strongly reconmend you to check your `ROS_PACKAGE_PATH`. `export ORB_SLAM_PATH={your orb_slam2 path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2` `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:${ORB_SLAM_PATH}` run command `cd ${ORB_SLAM_PATH}` in terminal to insure the path is correct. and then you can run `./build_ros.sh`
> > > 
> > > 
> > > My result is this. echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS How can I remove the medium part ? (/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS)
> > 
> > 
> > You can redefine the variable: ` export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS`
> 
> I try your method.
> 
> echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS
> 
> However, It still have the problem.
> 
> CMake Error at /opt/ros/melodic/share/ros/core/rosbuild/private.cmake:99 (message): [rosbuild] rospack found package "ORB_SLAM2" at "", but the current directory is "/home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2". You should double-check your ROS_PACKAGE_PATH to ensure that packages are found in the correct precedence order.

Maybe you should try  `/ORB_SLAM2/Examples/ROS/ORB_SLAM2` instead of `/ORB_SLAM2/Examples/ROS/`

nhjydywd, 2022-05-18:
> I try your method.
>
> echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS
>
> However, It still have the problem.
>
> CMake Error at [...]

`${ORB_SLAM_PATH}` should be configured as the path of `ROS` directory in `ORB_SLAM2` . Can you access that directory through `cd ${ORB_SLAM_PATH}`?

tungtungyan, 2022-05-19:
> `${ORB_SLAM_PATH}` should be configured as the path of `ROS` directory in `ORB_SLAM2` . Can you access that directory through `cd ${ORB_SLAM_PATH}`?

I try /ORB_SLAM2/Examples/ROS/ORB_SLAM2, but it is still an error.  How can I access that directory through cd ${ORB_SLAM_PATH}?

nhjydywd, 2022-05-19:
> I try /ORB_SLAM2/Examples/ROS/ORB_SLAM2, but it is still an error. How can I access that directory through cd ${ORB_SLAM_PATH}?

It's a way to check the enviromental variable.Type this command in the terminal:  `cd ${ORB_SLAM_PATH}`,  then your terminal are supposed to enter that directory. if your terminal didn't, it means that there is something wrong with your  ` ${ORB_SLAM_PATH}`.

tungtungyan, 2022-05-19:
> It's a way to check the enviromental variable.Type this command in the terminal: `cd ${ORB_SLAM_PATH}`, then your terminal are supposed to enter that directory. if your terminal didn't, it means [...]

I tried cd ${ORB_SLAM_PATH}, but it did not have any directory. How can I set ${ORB_SLAM_PATH}? Many thanks 🙏

nhjydywd, 2022-05-19:
> I tried cd ${ORB_SLAM_PATH}, but it did not have any directory. How can I set ${ORB_SLAM_PATH}? Many thanks 🙏

`export ORB_SLAM_PATH={path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2` , where `｛path}` is the absolute path of source code folder of ` ORB_SLAM2 `

tungtungyan, 2022-05-19:
> `export ORB_SLAM_PATH={path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2` , where `｛path}` is the absolute path of source code folder of `ORB_SLAM2`

I try it but still have the problem. 

In ~/.bashrc, I add the two line.
export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2
export ORB_SLAM_PATH=/home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2

echo $ORB_SLAM_PATH 
/home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2
echo $ROS_PACKAGE_PATH
/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2

However, I deleted all the export lines in  ~/.bashrc.
The result of echo $ROS_PACKAGE_PATH, show as below:
/opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS

I don't know how to delete it.

nhjydywd, 2022-05-19:
> I try it but still have the problem.
>
> In ~/.bashrc, I add the two line. export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2 export [...]

I remember that `/opt/source/melodic/setup.bash ` will also export some enviromental variables， maybe you can check that.  I‘m not sure because I have uninstalled ROS for a long time😬

tungtungyan, 2022-05-25:
I had solved by  sudo ln -s ~/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2 /opt/ros/melodic/share/ORB_SLAM2

tungtungyan, 2022-05-25:
> > > > > > > > > > > I just solved the same problem as you encountered. The cause of the problem is that my `ROS_PACKAGE_PATH` is wrong. (Although I once thought it was correct) So I strongly reconmend you to check your `ROS_PACKAGE_PATH`. `export ORB_SLAM_PATH={your orb_slam2 path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2` `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:${ORB_SLAM_PATH}` run command `cd ${ORB_SLAM_PATH}` in terminal to insure the path is correct. and then you can run `./build_ros.sh`
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > My result is this. echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS How can I remove the medium part ? (/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS)
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > You can redefine the variable: ` export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS`
> > > > > > > > 
> > > > > > > > 
> > > > > > > > I try your method.
> > > > > > > > echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS
> > > > > > > > However, It still have the problem.
> > > > > > > > CMake Error at /opt/ros/melodic/share/ros/core/rosbuild/private.cmake:99 (message): [rosbuild] rospack found package "ORB_SLAM2" at "", but the current directory is "/home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2". You should double-check your ROS_PACKAGE_PATH to ensure that packages are found in the correct precedence order.
> > > > > > > 
> > > > > > > 
> > > > > > > `${ORB_SLAM_PATH}` should be configured as the path of `ROS` directory in `ORB_SLAM2` . Can you access that directory through `cd ${ORB_SLAM_PATH}`?
> > > > > > 
> > > > > > 
> > > > > > I try /ORB_SLAM2/Examples/ROS/ORB_SLAM2, but it is still an error. How can I access that directory through cd ${ORB_SLAM_PATH}?
> > > > > 
> > > > > 
> > > > > It's a way to check the enviromental variable.Type this command in the terminal: `cd ${ORB_SLAM_PATH}`, then your terminal are supposed to enter that directory. if your terminal didn't, it means that there is something wrong with your ` ${ORB_SLAM_PATH}`.
> > > > 
> > > > 
> > > > I tried cd ORBSLAMPATH,butitdidnothaveanydirectory.HowcanIset{ORB_SLAM_PATH}? Many thanks 🙏
> > > 
> > > 
> > > `export ORB_SLAM_PATH={path}/ORB_SLAM2/Examples/ROS/ORB_SLAM2` , where `｛path}` is the absolute path of source code folder of `ORB_SLAM2`
> > 
> > 
> > I try it but still have the problem.
> > In ~/.bashrc, I add the two line. export ROS_PACKAGE_PATH=/opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2 export ORB_SLAM_PATH=/home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2
> > echo $ORB_SLAM_PATH /home/ytl/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2 echo $ROS_PACKAGE_PATH /opt/ros/melodic/share:PATH/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2
> > However, I deleted all the export lines in ~/.bashrc. The result of echo $ROS_PACKAGE_PATH, show as below: /opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS
> > I don't know how to delete it.
> 
> I remember that `/opt/ros/melodic/setup.bash ` will also export some enviromental variables， maybe you can check that. I‘m not sure because I have uninstalled ROS for a long timegrimacing

I had solved by  sudo ln -s ~/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2 /opt/ros/melodic/share/ORB_SLAM2

Mr-brillianter, 2022-09-21:
Things get strange. i also encountered this problems but it turned out that roslaunch could auto-complete with TAB, but ROS also could not find it. I also echo the variable and find the right path exists . I am trying to use catkin_make.