raulmur/ORB_SLAM2 #605: Segmentation Fault (Core Dumped) when running any examples

JoeyDow, 2018-07-03:
I am new to ORBSLAM2 and have been trying to troubleshoot from the previous posts, but none of them seem to resolve the problem. 
When i try and run any example (tried both EUROC and TUM datasets) i get a segmentation fault (core dumped). I am running Ubuntu 16.04LTE

Output from ORBSLAM2:
`ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

-------
Start processing sequence ...
Images in the sequence: 792

New map created with 832 points
Segmentation fault (core dumped)`

And with GDB on the output from bt is:

`
Depth Threshold (Close/Far Points): 3.09294
[New Thread 0x7fffd4891700 (LWP 21137)]
[New Thread 0x7fffd4090700 (LWP 21138)]
[New Thread 0x7fffd388f700 (LWP 21139)]

-------
Start processing sequence ...
Images in the sequence: 792

[New Thread 0x7fffd1ba1700 (LWP 21140)]
[New Thread 0x7fffd13a0700 (LWP 21141)]
[New Thread 0x7fffd0b9f700 (LWP 21142)]
[New Thread 0x7fffc3fff700 (LWP 21143)]
[New Thread 0x7fffc37fe700 (LWP 21144)]
[New Thread 0x7fffc2ffd700 (LWP 21145)]
[New Thread 0x7fffc27fc700 (LWP 21146)]
New map created with 834 points
[New Thread 0x7fffa7fff700 (LWP 21147)]
[New Thread 0x7fffa77fe700 (LWP 21148)]
[New Thread 0x7fffa6ffd700 (LWP 21149)]

Thread 1 "rgbd_tum" received signal SIGSEGV, Segmentation fault.
0x00007ffff51ffcc5 in g2o::EdgeSE3ProjectXYZOnlyPose::linearizeOplus() ()
   from /home/joe/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
(gdb) bt
#0  0x00007ffff51ffcc5 in g2o::EdgeSE3ProjectXYZOnlyPose::linearizeOplus() ()
   from /home/joe/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
#1  0x00007ffff7b6096c in g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::buildSystem() () from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#2  0x00007ffff5246998 in g2o::OptimizationAlgorithmLevenberg::solve(int, bool)
    () from /home/joe/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
#3  0x00007ffff523dd17 in g2o::SparseOptimizer::optimize(int, bool) ()
   from /home/joe/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
#4  0x00007ffff7b44aab in ORB_SLAM2::Optimizer::PoseOptimization(ORB_SLAM2::Frame*) () from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#5  0x00007ffff7ae14af in ORB_SLAM2::Tracking::TrackReferenceKeyFrame() ()
   from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#6  0x00007ffff7ae7770 in ORB_SLAM2::Tracking::Track() ()
   from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#7  0x00007ffff7ae8eae in ORB_SLAM2::Tracking::GrabImageRGBD(cv::Mat const&, cv::Mat const&, double const&) () from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#8  0x00007ffff7ad2e28 in ORB_SLAM2::System::TrackRGBD(cv::Mat const&, cv::Mat const&, double const&) () from /home/joe/ORB_SLAM2/lib/libORB_SLAM2.so
#9  0x0000000000407e0b in main ()
(gdb) 
`
Thanks for any help, this has me stumped

syywh, 2018-10-11:
Have you solved this problem? I have met the same problem recently. But when I work in 14.04 everything is OK...

azuryl, 2024-07-23:
I meet the same issue in https://github.com/MisEty/RTG-SLAM/issues/28

azuryl, 2024-07-23:
in  CMakeLists.txt
#set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE}")
SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native") 
#set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE}")
SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native")

ccwss007, 2024-09-04:
If I use the following command, a new error will occur: `SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native")`  
new error     ：   __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: 没有那个文件或目录.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737098502016)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737098502016, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff4a42476 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#4  0x00007ffff4a287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff4a89676 in __libc_message
    (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4bdbb77 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff4aa0cfc in malloc_printerr
    (str=str@entry=0x7ffff4bde790 "double free or corruption (out)")
    at ./malloc/malloc.c:5664
#7  0x00007ffff4aa2e70 in _int_free
    (av=0x7ffff4c1ac80 <main_arena>, p=0x555563db52f0, have_lock=<optimized out>) at ./malloc/malloc.c:4588
#8  0x00007ffff4aa5453 in __GI___libc_free (mem=<optimized out>)
    at ./malloc/malloc.c:3391
#9  0x00007ffff772955a in g2o::HyperGraph::clear() ()
    at /home/lyj/YOLO_ORB_SLAM3_with_pointcloud_map-master/Thirdparty/g2o/lib/libg2o.so

ccwss007, 2024-09-04:
If I use the following command, a new error will occur: SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native")
new error ： __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016) at ./nptl/pthread_kill.c:44
44 ./nptl/pthread_kill.c: 没有那个文件或目录.
(gdb) bt
#0 __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016)
at ./nptl/pthread_kill.c:44
https://github.com/raulmur/ORB_SLAM2/issues/1 __pthread_kill_internal (signo=6, threadid=140737098502016)
at ./nptl/pthread_kill.c:78
https://github.com/raulmur/ORB_SLAM2/pull/2 __GI___pthread_kill (threadid=140737098502016, signo=signo@entry=6)
at ./nptl/pthread_kill.c:89
https://github.com/raulmur/ORB_SLAM2/issues/3 0x00007ffff4a42476 in __GI_raise (sig=sig@entry=6)
at ../sysdeps/posix/raise.c:26
https://github.com/raulmur/ORB_SLAM2/issues/4 0x00007ffff4a287f3 in __GI_abort () at ./stdlib/abort.c:79
https://github.com/raulmur/ORB_SLAM2/issues/5 0x00007ffff4a89676 in __libc_message
(action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4bdbb77 "%s\n")
at ../sysdeps/posix/libc_fatal.c:155
https://github.com/raulmur/ORB_SLAM2/issues/6 0x00007ffff4aa0cfc in malloc_printerr
(str=str@entry=0x7ffff4bde790 "double free or corruption (out)")
at ./malloc/malloc.c:5664
https://github.com/raulmur/ORB_SLAM2/issues/7 0x00007ffff4aa2e70 in _int_free
(av=0x7ffff4c1ac80 <main_arena>, p=0x555563db52f0, have_lock=) at ./malloc/malloc.c:4588
https://github.com/raulmur/ORB_SLAM2/issues/8 0x00007ffff4aa5453 in __GI___libc_free (mem=)
at ./malloc/malloc.c:3391
https://github.com/raulmur/ORB_SLAM2/issues/9 0x00007ffff772955a in g2o::HyperGraph::clear() ()

ccwss007, 2024-09-04:
> in CMakeLists.txt #set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE}") SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native") #set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE}") SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native")

If I use the following command, a new error will occur: SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native")
new error ： __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016) at ./nptl/pthread_kill.c:44
44 ./nptl/pthread_kill.c: 没有那个文件或目录.
(gdb) bt
#0 __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737098502016)
at ./nptl/pthread_kill.c:44
https://github.com/raulmur/ORB_SLAM2/issues/1 __pthread_kill_internal (signo=6, threadid=140737098502016)
at ./nptl/pthread_kill.c:78
https://github.com/raulmur/ORB_SLAM2/pull/2 __GI___pthread_kill (threadid=140737098502016, signo=signo@entry=6)
at ./nptl/pthread_kill.c:89
https://github.com/raulmur/ORB_SLAM2/issues/3 0x00007ffff4a42476 in __GI_raise (sig=sig@entry=6)
at ../sysdeps/posix/raise.c:26
https://github.com/raulmur/ORB_SLAM2/issues/4 0x00007ffff4a287f3 in __GI_abort () at ./stdlib/abort.c:79
https://github.com/raulmur/ORB_SLAM2/issues/5 0x00007ffff4a89676 in __libc_message
(action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4bdbb77 "%s\n")
at ../sysdeps/posix/libc_fatal.c:155
https://github.com/raulmur/ORB_SLAM2/issues/6 0x00007ffff4aa0cfc in malloc_printerr
(str=str@entry=0x7ffff4bde790 "double free or corruption (out)")
at ./malloc/malloc.c:5664
https://github.com/raulmur/ORB_SLAM2/issues/7 0x00007ffff4aa2e70 in _int_free
(av=0x7ffff4c1ac80 <main_arena>, p=0x555563db52f0, have_lock=) at ./malloc/malloc.c:4588
https://github.com/raulmur/ORB_SLAM2/issues/8 0x00007ffff4aa5453 in __GI___libc_free (mem=)
at ./malloc/malloc.c:3391
https://github.com/raulmur/ORB_SLAM2/issues/9 0x00007ffff772955a in g2o::HyperGraph::clear() ()

hplegend, 2024-12-14:
I solve the problem by re-compile the g2o in the Thirdparty directory