raulmur/ORB_SLAM2 #611: Starting ROS node for Monocular camera

ujjwalrathod007, 2018-07-08:
I have multiole problems while processing live input of Monocular camera using ROS and SLAM2 library
1st 
I always have to export ORB_SLAM2 path when I stsrt a new session of bash using export command then only I can find ORB-SLAM2 package.(see the code below)

`ujjval@ujjval-VPCEH18FG:~$ rospack find ORB_SLAM2
[rospack] Error: package 'ORB_SLAM2' not found
ujjval@ujjval-VPCEH18FG:~$ export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/ujjval/ORB_SLAM2
ujjval@ujjval-VPCEH18FG:~$ rospack find ORB_SLAM2/home/ujjval/ORB_SLAM2/Examples/ROS/ORB_SLAM2
`
2nd
when I do rosrun with Vocabulary and settings file provided I don't get any output until I press (Ctrl+X),(Ctrl+X) and (Cntr+C) then the slam starts with Waiting for images message but no camera video. 

`ujjval@ujjval-VPCEH18FG:~$ rosrun ORB_SLAM2 Mono /home/ujjval/ORB_SLAM2/Vocabulary/ORBvoc.txt  /home/ujjval/ORB_SLAM2/Examples/Monocular/TUM1.yaml
[ERROR] [1531064871.417644383]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
^X^X^C
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
`
Anyone has an idea what could be the problem?

ArifSohaib, 2018-07-09:
Yes. I just installed it tonight and I am facing the second issue.
I was thinking it might need one of the mentioned datasets to load the images but I haven't tried it yet.
For the first one you just need to add the export command to the end of your .bashrc file.

ujjwalrathod007, 2018-09-08:
For Stereo I saw the example that it needs one of the ROS bag files. but for Mono I don t think it is so. Though I tried providing the cammand with image sequence but it did't seem to work.

pankhurivanjani, 2019-01-07:
I am also facing this similar issue.. did anyone solved it?

valarsundarlis, 2019-12-04:
I have found that adding the `ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}` line as the last line in ~./bashrc works, as it sources properly in a new terminal.