raulmur/ORB_SLAM2 #617: Pangolin x11: unable to retrieve framebuffer options

xuwh15, 2018-07-15:
@raulmur I am trying to run the ORB-SLAM2 code with TUM dataset on a ubuntu 16.04 google cloud VM. While performing this command:
`./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml PATH_TO_SEQUENCE_FOLDER` 
I got this error:
[image]
I tried to fix it according to 
stevenlovegrove/Pangolin#74 (comment)
but the new Pangolin version has no GLX_SAMPLE_BUFFERS and GLX_SAMPLES defined in the code now.
So how can I solve my problem now?

mromanelli, 2018-07-24:
I followed the procedure described here https://github.com/stevenlovegrove/Pangolin/issues/260#issuecomment-386968496 and it worked.

dnwls, 2019-05-02:
https://woojjang.tistory.com/52

NickPerezCarletonUniversity, 2021-06-30:
how do people find solutions to these problems? It kind of blows my mind... what tools would I use for debugging Pangolin?

NickPerezCarletonUniversity, 2021-06-30:
also, after I make changed to Pangolin, do I just go into my Pangolin build directory and type:

cmake ..
cmake --build .

or do I have to delete something first?... Do i have to rebuild Orb Slam as well?

AnhTuDo1998, 2021-07-02:
@NickPerezCarletonUniversity Since Pangolin is a depedency of ORB-SLAM, it is safe to rebuild ORB-SLAM2 as well. Meaning you will have to build the modified Pangolin, then run the build.sh script again for ORB-SLAM2

For my case, where I was cross-compiling Pangolin for an embedded platform, I quickly learnt that the graphic libraries are really a pain to deal with and just decide to switch off Pangolin usage from the orb-slam2 itself. If you are working towards ORB-SLAM2 on ROS, i suggest finding some wrapper repository on the internet that can allow you to stream the trajectory to a trajectory server, such as hector-trajectory-server and then use RVIZ to visualize it. The default usage of Pangolin might be quite inflexible in my opinion.

You can check out this [wrapper for ORB-SLAM3](https://github.com/thien94/orb_slam3_ros_wrapper) which is pretty neat, pretty sure there has been some similar work with ORB-SLAM2 as well

NickPerezCarletonUniversity, 2021-07-12:
@xuwh15 It is apparently a well known problem. OpenGL won't work over remote desktop. I haven't found any solutions to it.

NickPerezCarletonUniversity, 2021-07-12:
@AnhTuDo1998 do you have a repository you can upload showing your changes? and also a brief tutorial on how to set it up? I see I am not the only one who has to navigate around using Pangolin. did you try that orb-slam3 ROS wrapper? I wonder if it is buggy at all... I've noticed a few bugs people have mentioned in the main Orb-Slam3 repo.

zoldaten, 2022-11-15:
`nano ~/Pangolin/components/pango_windowing/src/display_x11.cpp`
correct as in https://woojjang.tistory.com/52 pointed out (right spelling below):
`GLXFBConfig* fbc = glXGetFBConfigs(display, DefaultScreen(display), &fbcount);`
and
`//throw std::runtime_error("Pangolin X11: Invalid GLX version. Require GLX >= 1.3");`

no need GLX_SAMPLE_BUFFERS change.
rebuild ONLY Pangolin not ORB_SLAM:
```
cmake -B build -GNinja
cmake --build build
```