raulmur/ORB_SLAM2 #633: KeyFrameTrajectory is empty

ravindraji4, 2018-08-14:
Hi all, 
I processed the Malaga dataset with left camera images. 
It worked fine and the result looks like the following:
[image]

But there are no values stored in the **"KeyFrameTracjectory.txt"** file. **It is empty**.
Could anybody please suggest a solution. 
Thank you very much in advance.

sberryman, 2018-08-15:
I'm new to ORB-SLAM but based on the screen shot it doesn't look like a good initialization has occurred so you don't have a valid map. So it is saving an empty file because no cameras could be localized.

Maybe try and bump up the number of features by a few orders of magnitude. Also you are showing a camera with zero distortion, are you sure this camera has zero distortion? What type of movement are you doing in the video? You'll need to translate the camera (not just rotate it) to get a good initialization. What does the subject and background look like, is there enough texture for it to find sufficient feature points across the entire image?

Again, I'm new to SLAM and SfM so these are just a few comments based on very little experience and reading.

ravindraji4, 2018-08-17:
Thanks for the comment Sberryman. 
The dataset I am using is a standard Malaga dataset number 7. The distorted images are rectified in this dataset and camera calibration are provided.

RahmanMifta, 2018-09-12:
@ravindraji4 I am also getting empty "KeyFrameTracjectory.txt". Have you found any solution to this problem? please, share the solution if you have.

Yousra-Regaya, 2024-12-03:
Hi
Were you able to generate the trajectories for the Malaga dataset sequences? if yes would you share the data with me as I am not even able to make the code work for the provided KITTI Dataset.

supermice, 2024-12-03:
邮件已收到！ ***@***.***

lx-r, 2024-12-03:
您好，来信已收到，谢谢！