raulmur/ORB_SLAM2 #634: Segmentation Fault because of Global Bundle Adjustment

ashok-y, 2018-08-15:
I am running a modified version of ORBSLAM2 where I can save the map features to a file and can reload the map. I'm running into an issue where if I reload the map features and there's a Loop Closure in the dataset, Global Bundle Adjustments occurs and at that point I'm receiving a Segmentation Fault in the g2o library. 
gdb Trace Log:
```
#0  0x00007ffff1ee055b in g2o::HyperGraph::addEdge(g2o::HyperGraph::Edge*) ()
   from /ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so

#1  0x00007ffff1eec9ba in g2o::OptimizableGraph::addEdge(g2o::HyperGraph::Edge*) ()
   from /ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so

#2  0x00007ffff7a67450 in ORB_SLAM2::Optimizer::BundleAdjustment (
    vpKFs=std::vector of length 1553, capacity 1553 = {...}, 
    vpMP=std::vector of length 175862, capacity 175862 = {...}, 
    nIterations=nIterations@entry=10, pbStopFlag=pbStopFlag@entry=0x73f2e4a, 
    nLoopKF=nLoopKF@entry=1553, bRobust=bRobust@entry=false)
    at /ORB_SLAM2/src/Optimizer.cc:172

#3  0x00007ffff7a67d47 in ORB_SLAM2::Optimizer::GlobalBundleAdjustemnt (
    pMap=0x15a62160, nIterations=nIterations@entry=10, 
    pbStopFlag=pbStopFlag@entry=0x73f2e4a, nLoopKF=nLoopKF@entry=1553, 
    bRobust=bRobust@entry=false)
    at /ORB_SLAM2/src/Optimizer.cc:45

#4  0x00007ffff7a013b2 in ORB_SLAM2::LoopClosing::RunGlobalBundleAdjustment (
    this=0x73f2c40, nLoopKF=1553)
    at ORB_SLAM2/src/LoopClosing.cc:680

#5  0x00007ffff2ffb8f0 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6

#6  0x00007ffff27466ba in start_thread (arg=0x7fff55318700)
```
Any input is appreciated. 
Thanks in advance.

mcallistertyler95, 2019-03-14:
Did you manage to fix this? I am having the same issue.

hnsywangxin, 2019-11-28:
@mcallistertyler95 @ashok-y @raulmur
I have the same issue,expect your help,thank you !

bhaveshachhada, 2020-02-27:
I am also having segmentation issue while using modified ORBSLAM2 code for map save-load functionality, but program exists just after getting enough keypoints for initialization. ThankYou.
@raulmur @mcallistertyler95 @ashok-y