raulmur/ORB_SLAM2 #639: Can't success build in Ubuntu18.04

zxc1234529, 2018-08-21:
I can build ORB_SLAM2 in Ubuntu16.04.
But the same method can't work in Ubuntu18.04.
Does ORB_SLAM2 support Ubuntu18.04? 
Or my method is wrong?
Thanks a lot.

anliec, 2018-08-22:
Not sure it will help you, but I have a modified version that I managed to compile on Ubuntu 18.04 ([https://github.com/anliec/ORB_SLAM2](https://github.com/anliec/ORB_SLAM2)).
The main change are C++11 support and OpenCV 4.0 compatibility.
I must admit that the change I made are a bit dirty (I probably broke compatibility with c++ lower than 11 and OpenCV lower than 3.4, even 4.0 maybe) but I hope it can help you.
If you use the master branch of (OpenCV)[https://github.com/opencv/opencv] you should be fine (If you forget the fact that it's not a stable release...)

zxc1234529, 2018-08-23:
@anliec this helps me a lot solved my problem perfectly. Thanks a lot!!!

closed 2018-08-25 by zxc1234529