raulmur/ORB_SLAM2 #640: Track failed easily

wang735249037, 2018-08-22:
Hi:
  could you help me? I am trying to use ORBSLAM to process my video, but it failed easily, do you have some advise on how to find the real reason caused the tracking lost?
  the images i used have many similarscene, and is textureless (still it can extract more than 4000 points per image,but seems the descriptor are very similar). i also saved the matched feature points on images in trackingwithmotionmodel in tracking.cc, and createnewmappoints in localmapping.cc, but seems the matched feature points are mostly correct... I also want to show the new map points in tracklocalmap, but i have no good idea on how to know wheather the feature points are correctly matched.
 Thank you in advance.

xiangweifeng, 2018-11-22:
Do u have solve the problem. I face same problem.

wang735249037, 2018-11-24:
no, I didn't got any comment, may be due to lack of video, which is forbidden to upload by my adviser.
I am trying to debug to see if any reason can be found.