raulmur/ORB_SLAM2 #642: build_ros.sh gives an error

ujjwalrathod007, 2018-08-27:
I have encounterd rthe above error and tried to solve it with similar errors others have faced. but till now no clue how to solve this error. anyone knows?
`sudo ./build_ros.sh
[sudo] password for orbslam: 
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:4 (rosbuild_init):
  Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred!
See also "/opt/ros/kinetic/share/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
`

cmdraghici, 2018-09-01:
Got the same problem, the think is that if you use sudo you lose your defined vars. So change the perms to the root directory recursively so you wont get the perms errors when you build it without sudo.

ujjwalrathod007, 2018-09-05:
I have done the following

> sudo chown -R myusername:root ORB_SLAM2

then I go in ORB directory and run

> ./build_ros.sh

`Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
[rosbuild] Building package ORB_SLAM2
Failed to invoke /opt/ros/kinetic/bin/rospack deps-manifests ORB_SLAM2
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 130, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 633, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 533, in load_cached_sources_list
    return parse_sources_data(cache_data, origin=cache_index, model=model)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 388, in parse_sources_data
    raise InvalidData('line:\n\t%s\n%s' % (line, e), origin=origin)
rosdep2.core.InvalidData: line:
	yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4
[rospack] Error: could not call python function 'rosdep2.rospack.init_rospack_interface'

CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM2'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/opt/ros/kinetic/share/ORB_SLAM2/Examples/ROS/ORB_SLAM2/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
`

gnykumar, 2019-01-15:
Hi.
Same error i faced
Ubuntu 16 
Ros Kinetic

But it got resolved
by below

Add the path including Examples/ROS/ORB_SLAM2 to the ROS_PACKAGE_PATH environment variable. Open .bashrc file and add at the end the following line. Replace PATH by the folder where you cloned ORB_SLAM2:

kunnalparihar, 2019-04-14:
I faced the same error in Kinetic 
It is working now by adding the following line in ~/.bashrc file. Replace PATH by the folder where you cloned ORB_SLAM2.
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM2/Examples/ROS

then run
$ source ~/.bashrc

check that 
$ echo $ROS_PACKAGE_PATH 
should give /opt/ros/kinetic/share:PATH/ORB_SLAM2/Examples/ROS

$echo $ROS_ROOT 
should give /opt/ros/kinetic/share/ros

daiwj5, 2019-04-17:
> I faced the same error in Kinetic
> It is working now by adding the following line in ~/.bashrc file. Replace PATH by the folder where you cloned ORB_SLAM2.
> export [...]

i have tried this method, but it still not worked  and gave following errors:
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/daiwj5/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is "/home/daiwj5/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2".
  You should double-check your ROS_PACKAGE_PATH to ensure that packages are
  found in the correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/daiwj5/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
how should i do?

ajay1606, 2019-06-27:
Hello All,

I am also getting the following error when I tried to build ORB_SLAM2 in ros. The error message is as follows.
```
Building ROS nodes
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /home/ajay/.pyenv/shims/python (found version "3.5") 
[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/ajay/orb_slam_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is "/home/ajay/orb_slam_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2".
  You should double-check your ROS_PACKAGE_PATH to ensure that packages are
  found in the correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)
-- Configuring incomplete, errors occurred!
See also "/home/ajay/orb_slam_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
```

Platform: Ubuntu 16.04
ROS : kinetic
Python version: 3.5

//FOLLOWED step
```
sudo gedit \.bashrc

export `ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/ajay/orb_slam_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2`

source ~/.bashrc
```

But still, I am unable to figure out the root of the issue. Could anyone please give me some suggestions.
Your help much appreciated

Thanks.
AJay

Keerthan020, 2019-06-28:
I had the same issue, it was solved when you add the line
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM2/Examples/ROS
at the end of  ~/.bashrc and not in between, hope this helped. :)

ajay1606, 2019-06-29:
@Keerthan020 Thank you so much for your kind response. Yes, it is working now.

IMHinnG, 2021-09-12:
It must be enter root the user to modify ,otherwise it is invalid  !!!!!!!!!! ,my system is Ubuntu 16.04.1

tungtungyan, 2022-05-18:
My result is this.
echo $ROS_PACKAGE_PATH
/opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS
How can I remove the medium part ? (/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS)

Aishkrish18, 2023-06-21:
./build_ros.sh 
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:4 (rosbuild_init):
  Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred!
See also "/home/aishwarya/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
I am still getting errors like this inspite of doing the above suggestions? Any thing that i need to do ?

lx-r, 2023-06-21:
您好，来信已收到，谢谢！