raulmur/ORB_SLAM2 #646: usleep function not found

cmdraghici, 2018-09-01:
Hi there,

I am having some big troubles there with this error if someone can help me would be great. Here is a part of he trace:
`In file included from /usr/local/include/eigen3/Eigen/Core:363:0,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/devel/lib/cmake/Pangolin/../../../include/pangolin/gl/gl.h:39,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/devel/lib/cmake/Pangolin/../../../include/pangolin/pangolin.h:33,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/include/MapDrawer.h:27,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/include/Viewer.h:26,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/include/Tracking.h:28,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/include/System.h:29,
                 from /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:23:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackStereo(const cv::Mat&, const cv::Mat&, const double&)’:
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:134:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackRGBD(const cv::Mat&, const cv::Mat&, const double&)’:
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:185:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackMonocular(const cv::Mat&, const double&)’:
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:236:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc: In member function ‘void ORB_SLAM2::System::Shutdown()’:
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:309:24: error: ‘usleep’ was not declared in this scope
             usleep(5000);
                        ^
/home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin/orb_slam_2_src-prefix/src/orb_slam_2_src/src/System.cc:315:20: error: ‘usleep’ was not declared in this scope
         usleep(5000);
                    ^
make[5]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[4]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [orb_slam_2_src-prefix/src/orb_slam_2_src-stamp/orb_slam_2_src-build] Error 2
make[1]: *** [CMakeFiles/orb_slam_2_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/cezar/ros_ws/ORB-SLAM2-ROS/build/orb_slam_2_catkin; catkin build --get-env orb_slam_2_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << orb_slam_2_catkin:make               [ Exited with code 2 ]      
Failed    <<< orb_slam_2_catkin                    [ 2 minutes and 35.3 seconds ]
Abandoned <<< orb_slam_2_ros                       [ Unrelated job failed ]    
[build] Summary: 4 of 6 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3 minutes and 34.0 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`
Can't put the complete trace because is too large.
Thanks in advice.

aixi, 2019-01-13:
include <unistd.h> works for me

chijinxina, 2019-05-15:
include <unistd.h>   in  system.h