raulmur/ORB_SLAM2 #649: Starting ROS Mono node

ujjwalrathod007, 2018-09-08:
I have some problem starting the Mono node from the  SLAM2.
when I do

    rosrun ORB_SLAM2 Mono <path_to_vocabulary_file> <path_to_yaml_file>

it starts and loads the vocabulary but then the frame window remains empty and displays 'waiting for the images'. So whats the method to start Monocular node from topic **/camera/image_raw?**

cecileclav, 2018-09-22:
Hi, 
you need to play a rosbag of the sequence you want to run the program on, in an other terminal window. It is fully explained on the README, at the end, but I copy the lines here : 

Download a rosbag (e.g. V1_01_easy.bag) from the EuRoC dataset (http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). Open 3 tabs on the terminal and run the following command at each tab:

roscore

rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
(that's what you already do, but you need to open a new terminal window and enter the following command line)

rosbag play --pause V1_01_easy.bag /cam0/image_raw:=/camera/left/image_raw /cam1/image_raw:=/camera/right/image_raw

Then, you just have to tap on the space tab to run the rosbag and ORB-SLAM2 will be provided with the images of the rosbag.

ujjwalrathod007, 2018-09-23:
@cecileclav Yes in case of sterio there was no problem running the node.. Actually in case of Mono when I run 
rosbag play --pause /home/orbslam/myownbag.bag /cam0/image_raw:=/camera/image_raw

I saw there was nothing!! 
Then I tried to publish images from topic camera/image_raw to record it using rosbag record topic but that didnot work. 
I want to process my own data and for that I need to create .bag files.

BW25, 2018-11-02:
ORBSLAM is subscribing to the Ros topic camera/image_raw, but nothing is being published there.
So it is looking at nothing.
We need to publish something to that topic.

This can be a .bag file, like with the stereo example.
We can also use live input from a camera.

There are a number of Ros nodes that can do this, I tried cv_camera.
cv_camera publishes to a different topic, but we can remap it to publish to /camera/image_raw by adding this argument when running it:
 /cv_camera/image_raw:=/camera/image_raw

Full disclosure, I only just got this working myself, and ORBSLAM2 still isn't initializing, but hopefully this will help put you on the right track.

tnaderi, 2019-02-26:
Hi, 
I have the same issue. Where should I put "/cv_camera/image_raw:=/camera/image_raw" in which file?
Thanks

BW25, 2019-02-26:
See this link on remapping arguments in ROS.
http://wiki.ros.org/Remapping%20Arguments

The remapping argument is given when you run cv_camera.
Add it to the command you use to start the cv_camera node.

tnaderi, 2019-02-26:
Thank you for your response. That is very clarifying. When I used calibration http://wiki.ros.org/camera_calibration the format of file is different from format of yaml file you have used in your orb_slam2 config folder. How can I convert the two format to each other. I tried to do that but the map is highly distorted. Is it possible that my calibration has problem?