raulmur/ORB_SLAM2 #652: Using a series of images for ORBSLAM2

BW25, 2018-09-18:
It seems that a typical use of ORBSLAM2 is to hook up a webcam and use the ROS node for live video.
I have a drone with a mounted camera that takes pictures at regular intervals.
Is it possible for ORBSLAM2 to process this in real time, and if so, how?

hudsonmartins, 2018-10-10:
Hi @BW25, I have to do the same. Did you managed to do it?

BW25, 2018-10-11:
No, not yet.
I go to it processing recorded video, but lack of a webcam and bugs have impeded progress in building the ROS node and testing it with either live video, or a stream of images.
Keep in mind though, the recorded video examples basically are just a stream of images.
Still, although I’ve heard some people managed it, I find ORBSLAM2 doesn’t work to well at a lower frame rate, which is what I really needed.

hudsonmartins, 2018-10-11:
I understand... Thanks for answering!

emredog, 2018-11-20:
You should check `mono_kitti.cc` or `mono_tum.cc` for this. Basically you need to save the timestamps for your images somewhere and implement your own `void LoadImages(...)` function to load the image paths and respective timestamps.