raulmur/ORB_SLAM2 #653: How to get pixel coordinates from Orb Feature in each frame?

Karth1kKumarK, 2018-09-21:
I am trying to obtain the 3d points only along the lane width, hence is there a way to get pixel coordinate from a 3d point for each frame, and filter out 3d points apart from those are  in the lane width

AlejandroSilvestri, 2018-10-17:
@Karth1kKumarK ,

I don't know where are you looking for that info.  I can't help you about trajectory files.

Each KeyFrame object has a vector of MapPoints (KeyFrame::mvpMapPoints), and a correspondent vector of KeyPoints (KeyFrame::mvKeys), so mappoint and keypoint under the same index belong to each other.

There is a KeyFrame::mDescriptors, an array of correspondent descriptors (a cv::Mat).

AnneCCollin, 2018-12-17:
I am in a similar situation, where I need the pixel coordinates of the descriptors for the stereo camera (specifically on KITTI). In the stereo example, is it possible to get the pixel coordinate of the descriptors for each of the right and left raw, rectified, images? (uR, uL, and v). It would also be nice to have a function that allows to go from a MapPoint to a KeyPoint and vice-versa, without having to go through the KeyFrame.

q412611827, 2019-01-04:
std::set<KeyFrame*> kf = mspKeyFrames(k);
cv::KeyPoint kp = kf->mvKeys[i];
int X = (int)kp.pt.x;
int Y = (int)kp.pt.y;

KeyFrames have KeyPoints.
KeyPoints have pixel coordinates.