raulmur/ORB_SLAM2 #655: Information matrix choices in Optimizer

DEQDON, 2018-09-28:
I hava a question about how the information matrices (inverse of covariance matrices) of reprojection error edges are chosen in the Optimizer.cc. 
The information matrices are set as identity matrices multiplied by inverse scale in the ovtave. An example is like this:

`
const float &invSigma2 = pKF->mvInvLevelSigma2[kpUn.octave];
`
`
e->setInformation(Eigen::Matrix2d::Identity()*invSigma2);
`

My question is, the reprojection errors are 2D pixel coordinates and seems to have no relation to "scale".
Could someone explain the rationale behind this? Or, should the information matrices simply be identity matrices?

AlejandroSilvestri, 2018-10-17:
@DEQDON 

KeyFrame's keypoints coordinates uses the full image as system of reference, no matter in which octave it was found.

So, the higher the octave, the lower the information.  It is like in higher octaves keypoint's pixels are bigger, their area are bigger, and the keypoint's position is less specific.  This is why information matrix are multiplied by invSigma2.

DEQDON, 2018-10-18:
Thanks! So can I interpret this is as a way of choosing different weights for keypoints in different octave levels.

AlejandroSilvestri, 2018-10-18:
@DEQDON 

Yes. that's right.  ORB-SLAM2 doesn't really involve information matrix, that's why it starts with identity.  Only because it is available, the information matrix has been chosen as a way to apply weight.

closed 2018-10-22 by DEQDON