raulmur/ORB_SLAM2 #657: Only orientation change is only seen

chrismath, 2018-10-02:
I am trying to use ORBSLAM2 for pose estimation using a stereo camera in ROS. I made the necessary correction in the setting file and I also added the orb vocabulary file also as mentioned. But in the ORB MAP viewer I can only see orientation change and not translation change could anyone help me out with this issue.

BW25, 2018-10-04:
I periodically get this issue myself.
I don't have a fix unfortunately, but I've noticed that after loop closing, the trajectory is corrected to show the translation change, and the issue goes away, so translation changes after the loop closing are shown.

AlejandroSilvestri, 2018-10-18:
My guess is that both of you had big scale drifting, so your current map shrunk and become diminute compared to the initial map.  Loop closing fix that.