raulmur/ORB_SLAM2 #660: ZED stereo camera with ORB SLAM on ROS

ArtlyStyles, 2018-10-09:
Hi, I am also thinking using ZED with ORB SLAM. I am wondering if I should use ROS to link ZED and ORB SLAM or write C++ code to integrate the two component directly? For my application, I hope to run the SLAM algorithm real time. I have ZED and Nvidia TX2 board on a "MIT RaceCar" (RC car) which is operated by ROS.  So using ROS to connect the two is easy for me. However, I noticed the ZED images through ROS node has a big latency, which may impact the real time performance.

mathuse, 2020-06-08:
> Hi, I am also thinking using ZED with ORB SLAM. I am wondering if I should use ROS to link ZED and ORB SLAM or write C++ code to integrate the two component directly? For my application, I hope to [...]

I have the stereo camera, and I migrate the code to my code. I find some problem,the first problem is that what time the program calculate the "pKF->mFeatVec" when program first enter the "TrackReferenceKeyFrame/SearchByBoW". I slove it by adding "mCurrentFrame->ComputeBoW();" before "StereoInitialization/new KeyFrame" but it is not perfect, because I have successfully tested stereo camera using the executable file compiled from ORB-SLAM2 code.
The second problem is "std::bad_alloc" when I run the my executing file, and I have find the location of this problem at "TrackReferenceKeyFrame/PoseOptimization/optimizer.optimize(its[it])". I want to know the reason.Can you help me? By the way, I have tried a lot of ways, but none of them worked.

shubhendu1907, 2020-11-23:
Hello @ArtlyStyles ,
How did you create the yaml file for ZED camera for calibration?