raulmur/ORB_SLAM2 #661: build.sh error

jinfagang, 2018-10-15:
Simply installed depencies and run build.sh. got error:

```
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^~~~~~~~~~
/home/jintain/work/codes/slam/ORB_SLAM2/src/Tracking.cc: In member function ‘void ORB_SLAM2::Tracking::Reset()’:
/home/jintain/work/codes/slam/ORB_SLAM2/src/Tracking.cc:1512:24: error: ‘usleep’ was not declared in this scope
             usleep(3000);
                        ^
CMakeFiles/ORB_SLAM2.dir/build.make:296: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/ORB_SLAM2.dir/build.make:88: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:101: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:75: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:62: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/System.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
CMakeFiles/Makefile2:257: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

It's this a bug or my mistake for missed something?

missminga, 2018-10-16:
my mistake is the same as yours,but the"EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80"is bug?

jinfagang, 2018-10-16:
Problem solved, thanks.

closed 2018-10-16 by jinfagang

plutomedia, 2019-05-19:
Hi. I tried the one above ('Put #include <unistd.h> in System.h.' from cbachhuber) but it didn't help. 
But: after reading the following post I tried to Downgrade the Eigen to 3.2 and all the problems magically dissappeared :)
https://github.com/raulmur/ORB_SLAM2/issues/317