raulmur/ORB_SLAM2 #666: How to extract numeric values for position/translation and rotation from pose matrix?

salocinx, 2018-10-23:
Hi. I successfully compiled ORB_SLAM2. It runs very well, but I can't manage to extract the position and rotation values from the pose matrix `cv::Mat tcw` received from `TrackStereo()` function.

I need the following information:

Position/tranlsation data:

```
double posX = ?
double posY = ?
double posZ = ?
```

Euler rotation data:

```
double rotYaw = ?
double rotPitch = ?
double rotRoll = ?
```

Quaternion would also be great. Could somebody point me to the right direction with some source code?

Thank you!

closed 2018-10-27 by salocinx

DodsGTH94, 2018-10-30:
Hi @salocinx ,
I write you because I did like this: I printed out on the terminal the matrix easily using cout function. How did you do? How are the positions you get, in terms of accuracy?
Thanks

salocinx, 2018-10-30:
Hi @DodsGTH94 , yes I found out about `std::cout << tcw << std::endl;` too 👍  You can access the elements of the matrix by doing so:

```
if(tcw.cols==4 && tcw.rows==4) {
    for(int y=0; y<4; y++) {
        for(int x=0; x<4; x++) {
            float element= tcw.at<float>(x, y);
            // do someting with field "element" ...
        }
    }
}
```

Didn't had the time to analyse accuracy and the interpreting the single elements yet.

What are the position data? The first row?
What are the rotation data? The second row?