raulmur/ORB_SLAM2 #668: disable pangoline view.

rahulsharma11, 2018-10-24:
Hi, i am using Orbslam2 on my rockpro64 board by ssh.
I also have made bool bUseViewer = false in System.h file.
But still getting error-
Start processing sequence ...
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to open X display
./stereo_live.sh: line 1: 19314 Aborted                 ./Examples/Stereo/stereo_live Vocabulary/ORBvoc.txt Examples/Stereo/KITTI00-02.yaml

Please suggest.
Thanks.

antoniovillavicencio, 2018-11-13:
You don't have to modify System.h, that's the definition of the file and you are breaking the program. Instead modify where you actually call that function. It's something like
`// Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM2::System SLAM(argv[1], argv[2], ORB_SLAM2::System::RGBD, false);`

That line is in your main programme.

almondboi, 2020-12-21:
> You don't have to modify System.h, that's the definition of the file and you are breaking the program. Instead modify where you actually call that function. It's something like
> `// Create SLAM [...]

In which file is this line situated in?

NickPerezCarletonUniversity, 2021-10-04:
i was able to get rid of the pangolin error by opening another terminal and typing in:
xhost +